pbatard/rufus #422: Rufus Doesn't Recognize USB

scottthomas121, 2015-01-02:
I have formatted the new USB drive at NTFS several times, but Rufus doesn't recognize the USB.

Rufus version 1.2.0.183
Found drive 'SanDisk Cruzer Glide USB Device'
0 devices found.

pbatard, 2015-01-02:
> Rufus version 1.2.0.183

The latest version is 1.4.12! You are using a very old version - please try the latest. You can download it at the official site: http://rufus.akeo.ie

If it still doesn't work, can you find out if the drive is listed in Windows Device Manager or Computer Management. If it doesn't appear there, it will never be listed in Rufus.

scottthomas121, 2015-01-04:
downloading a newer version helped!!!  Thanks for the assistance.

closed 2015-01-04 by pbatard