pbatard/rufus #427: Rufus does not see network drives on WIndows 7 64-bit

dtoubelis, 2015-01-08:
I was trying to burn and image from a network drive and rufus does not see it. The drive is mapped as S:. Also, when I try to search for network shares in the network it does not recognize any. What would it take to make it work?

pbatard, 2015-01-08:
Rufus runs as administrator, so it will see the same drives and files as an administrative user can see.
This means that if you mapped a drive as a standard user, just like you won't be able to navigate to it from an elevated command prompt, Rufus will not see it either. And of course if only a regular user has been granted access to network file, an admin app will not see it.

If you open a command prompt as administrator and try to access your files, or run any application, such as a text editor, as elevated, you will find the same limitations.

Unfortunately, Windows does not allow an application to switch user when running, which makes it exceedingly difficult to do anything about this. If you want to access any networked data, you will need to make sure that your Windows admin user can access it first.

Since there's not much I can do, I'll be closing this issue.

closed 2015-01-08 by pbatard

dtoubelis, 2015-01-08:
Oh I see. This totally makes sense. Thanks.