pbatard/rufus #430: Rufus 2.0.0.566 (Alpha) gui issue

azdps, 2015-01-12:
Rufus gui issue using Rufus 2.0.0.566 (Alpha)

I'm running Windows 7 x64

How to reproduce:
- start rufus
- select DD Image from drop down box
- click on "select and image button"
- select "all files" from the drop down list
- select a file other than a valid image file such as a text file
- rufus no longer has the gui controls restricted (darkened)

pbatard, 2015-01-12:
I appreciate the help, but please avoid reporting problems against __ALPHA__ releases.

ALPHA release are VERY UNFINISHED products, with bugs and features that are far from complete.
If everybody starts to report issues with the ALPHA, I'm going to have a list that is long like the arm in no time, and I'd rather concentrate on getting to BETA stage (where I'll accept issue reports) than have to answer a myriad of "Yes, I know it's not working __yet__" reports on an ALPHA.

closed 2015-01-12 by pbatard

pbatard, 2015-01-12:
Also, I should point out, regardless of controls being greyed out, Rufus will let you know that you cannot format if you used an invalid image, so this is a very minor problem, which I am probably never going to bother doing anything about unless I have a lot of time on my hands.

azdps, 2015-01-12:
This minor gui issue also applied to the latest build you have posted on your website. I just reported it against the ALPHA since the ALPHA build had the same issue. No worries, Rufus works very well and I appreciate the tool.