pbatard/rufus #433: Trojan warning on Metascan?

donnielrt, 2015-01-25:
Hey,

A trojan alert is shown for Rufus on Metascan, has this been confirmed to be a false positive?

https://www.metascan-online.com/en/scanresult/file/67fae016afd646fe8366dfe06074dee1

pbatard, 2015-01-26:
You know, I'm getting quite tired with these constant AV false positives...

About each release, I'll get a report, from a different AV solution (usually obtained from an automated online engine), stating the latest version might include malware. But then, every time I ask the AV vendors to do a __proper__ analysis, they confirm that the file is clean. This has happened about 6-7 times already, and enough is enough!

In this case, if __41__ AV vendors report the file to be clean, and one says it has an issue, then that one is most likely __wrong__.

So, from now on, I will only acknowledge a malware report that have been provided by __an actual employee from an AV vendor__, because it is clear that these automated meta engines are not helping.

therefore, if you feel that this report can be trusted (I don't), please contact filseclab and ask them to re-scan rufus.exe to confirm that it's a false positive. Then if, an only if, you get an employee from filseclab to indicate that the latest version appears to contains malware (and able to describe how this malware operates, so that it can be double checked), should you reopen this issue.

Also note that due to the nature of the compilation process, and the small size of the executable (which I monitor very carefully), it would be difficult to add malware to Rufus without being spotted, be it on my development machine or on the website.

closed 2015-01-26 by pbatard

Sopor-, 2015-01-26:
I can explain why @donnielrt get a Trojan alert. Rufus use an executable packer called UPX and when you compress a file they will look the same as Trojans and viruses because 99.9% of them are compressed with an executable packer. So if you unpack Rufus.exe with UPX and then run the file again against the virus checker you will not get this false positive. So if @pbatard stop using UPX all these false positives will disappear :smile:

pbatard, 2015-01-26:
UPS is a __legitimate__ compressor, just like zip, tar.gz or 7z.
By the same reasoning, should we stop using roads, because criminal use roads too?

Also this would inconvenience millions of people (because downloading Rufus would now take 3-4 times as long), and make the bandwidth usage from my server skyrocket.

Plenty of legitimate applications use UPX. If an AV solution thinks that there is something suspicious about using it, then you simply should not trust that solution, because it clearly has NO IDEA what is malware and what isn't.

Sopor-, 2015-01-26:
I only explain why Rufus is detected as a virus/trojan and if you want to avoid false positives there is a simple solution. @donnielrt can unpack it and then run a virus check.

pbatard, 2015-01-26:
But you have to understand that if everybody were to do what you advocate, the AV vendors would never fix what is purely a problem on their side.

As I stated, it not worth any developer's time to try to work around faulty AV solutions, because this actually doesn't help anyone, and especially not users, who, if software developers start to tiptoe around using UPX, will continue to get false positive on any executable that use it, or executables that they compressed themselves. In the end, everybody loses: AV vendors remain oblivious to the issue, and users continue to get false positives. That's not a solution at all.

donnielrt, 2015-01-26:
Hey @Sopor-, thanks for the explanation and suggestion! Good to know!

@pbatard hear ya, and you've got a complete fair point! If this was a paid app, I would've argued that the end user shouldn't care *why* there's an error, and it was the dev's responsibility to fix the problem.

HOWEVER, this is an open-source freeware app, and since you've previously already informed the antivirus vendors, I don't think there's anything more you *need* to do! 

Rufus seems to be a great app, thanks for taking the time to code and maintain it, and for dealing with annoying bug reports :)

pbatard, 2015-01-26:
@donnielrt - agreed (and thanks for the kind words).

If I hadn't been burnt doing so and felt like it is now a complete waste of time, I would indeed report the false positive to filseclab so that they can update their solution.

But as I tried to point out in my first post, I've been doing that for other AV solutions more times than I'd like already, and this is getting tiresome: it only takes a new security vendor being lazy and deciding that they've seen UPX being used a few too many times in malware, to flag UPX compression as a whole as a sign of malware, and require a repeat of this giant circus (NB: UPX is not the only thing from Rufus that has been falsely flagged as malware in the past - I've had this happen past decompression as well).

So all I am saying is: enough is enough. I tried to play the AV detection game, but that's a game you can't win as new (faulty) security solutions seem to pop up every other week. 

And because I'm providing the software for free, I will now require users to get confirmation from a human employee of the security vendor before I start investigate or apply a workaround for a report of possible malware...

donnielrt, 2015-01-26:
@pbatard :+1: Will also let the filseclab guys know about the false-positive ;)

pbatard, 2015-01-26:
Thanks! Much appreciated!!

kittrCZ, 2017-03-13:
I have seen couple of people coming to metadefender.com from this particular issue. Just to be clear, this was false positive and the file is clean as today: https://www.metadefender.com/#!/results/file/ZTE3MDMxM1NramdBeGR0RWpnU0ozbFJlZEtOc3g/regular/analysis (re-scanned on 2017-03-13 20:58:24 GMT)