pbatard/rufus #434: Rufus is great - but...

ueberlayman, 2015-01-30:
Rufus runs fine on my 2 GB Cruzer Blade when I start my HP NOTE-book. 

However, this same flash drive is not being recognised by my Packard Bell DOT S2 NET-book, despite the (hopefully) correct BIOS settings. 

Before Rufus, I had a different boot software on this flash drive made by some expert whom I can't find anymore, and whose (the software) name I don't really know (maybe Grub4DOS, but not sure). There, the flash drive was recognised by the netbook too, until one day, the software stopped working altogether - that's why I changed to Rufus.

Unless double-checking the netbook's BIOS settings makes any sense, I assume the problem is the flash drive. At least, I mean to have read that some other people have had similar problems. Did someone here have this same problem combination (Rufus - Cruzer Blade - DOT S2), and how did you solve it?

pbatard, 2015-02-01:
Could it be that your Packard Bell is using restrictive UEFI firmware that prevents MBR bootable USB Flash Drives from being recognized?

You will understand that the information you provide a bit tenuous ("some expert whom I can't find anymore, and whose (the software) name I don't really know" doesn't exactly help with getting data on how you may have gotten that UFD recognized before), and the fact that it works just fine with your other system seems to point to a system configuration issue or limitation, rather than a Rufus problem. There are plenty of __different__ ways to make USB drive bootable, that aren't interchangeable (they depend on what you are trying to boot: DOS will use a different boot process than Windows, which uses a different boot process from a Linux image, which in turn is different from Grub4DOS), so if your Packard Bell firmware is only compatible with a specific type of boot processes (say Grub4DOS) and not others, there's not much I'll be able to do in Rufus, as I can't really make everything switch to using Grub4DOS.

I would really advise you to spend time figuring out what exactly you used to make that Cruzer Blade recognized once, as without this data, there's not much I can do.

I'll probably leave this issue open for another week, but if nobody replies by then, I'll have to close it.

pbatard, 2020-10-05:
Forgot to close this issue.

closed 2020-10-05 by pbatard