pbatard/rufus #436: ER: Allow *SOME* cheat modes to be enabled permanently in ini/registry

pbatard, 2015-02-03:
Not sure if things that are voluntarily disabled such as listing USB HDDs should be part of that enhancement, but stuff like size units, VMWare VMHD listing or the upcoming timestamp preservation (#389), should probably be stored on permanent basis.

maxxc0m, 2015-03-03:
Self-assign

pbatard, 2015-03-03:
> Self-assign

???

MagicAndre1981, 2015-03-06:
this is a SPAM account.

But the option to always show USB HDDs would be nice.

pbatard, 2015-03-06:
For some. On the other hand, what wouldn't be nice is the angry reports I am virtually guaranteed to get from people who got that option enabled by default, and ended up formatting an external USB HDD with important data because they weren't paying attention, but will tell me how Rufus should have done much more to prevent that.

Heck, even with USB HDD detection never enabled by default, I got multiple complaints from people indicating that Rufus should have done more babysitting to prevent their data loss, when they picked up the wrong target. I did some of that by adding extra warnings when multiple partitions are detected on the target, but it never seems enough.

I'm afraid therefore that your convenience will have to come after my goal of preventing data loss. For that specific reason, besides the `-f` commandline switch (because it's difficult to leave that one enabled by mistake), I am not currently planning to have a setting that allows USB HDDs to be detected by default...

pbatard, 2015-08-04:
Delaying this to 2.4.

commit a9b68197 (2015-09-16): [ui] enable some settings and cheat modes to be persisted

* Currently, the user settings that are persisted are:
  - Advanced mode (UI)
  - Timestamps preservation (Alt-T)
  - Proper unit Display (Alt-U)
  - USB debug (Alt.)
  - Fake drives detection (Alt-B)
  - Dual UEFI+BIOS for Windows images (Alt-E)
  - Force Large FAT32 formatting (Alt-L)
  - VMWare Disk Detection (Alt-W)
* Settings that are *NOT* persisted include:
  - UDB HDD listing (too dangerous!)
  - NTFS compression (Alt-N)
  - ISO/Joliet/Rock-Ridge (too confusing if not reset)
  - Disabling of size limits (Alt-S)
* Closes #436
closed 2015-09-16 by pbatard