pbatard/rufus #437: ER: install.swm in place of .wim causes failure in uefi boot creation

Fuj-, 2015-02-04:
My install.wim is >4GB due to slipstreamed updates so I split it into .swm parts in order to be able to use FAT32. It causes Rufus to fail when setting up uefi boot loader because it tries to extract bootmgfw.efi from non-existent install.wim. 7z supports swm archives so it should be an easy fix, just check for install.wim/install.swm before running 7z.exe.

pbatard, 2015-02-05:
> My install.wim is >4GB due to slipstreamed updates so I split it into .swm parts in order to be able to use FAT32.

You do know that you aren't really constrained to use FAT32, and can use NTFS even for UEFI boot. As a matter of fact, the upcoming version of Rufus (Rufus 2.0, an ALPHA of which can be downloaded [here](https://rufus.akeo.ie/test/)) will happily let your format a Windows installation media containing a > 4GB file onto NTFS, and __seamlessly boot__ from NTFS on UEFI platforms (as long as they are x86_64, but I can add x86_32 or even ARM if needed).

I'd actually really appreciate if people could give it a try, because I spent a bit of time producing [EFI NTFS drivers](http://efi.akeo.ie/) as well as a method to seamlessly boot from NTFS using these drivers to, in part, solve the issue of booting Windows ISOs with a >4GB file in UEFI mode

This being said, I'll see what I can do to have Rufus work with `.swm`. I wasn't aware of this type of split archive beforehand, and find it quite silly that Microsoft went with `install.swm`, `install2.swm`, `Install3.swm` and so on, rather than `install.wim`, `install2.swm`, `Install3.swm`... Why they need to rely on extensions, rather than have the WIM header indicate that an archive is multipart is beyond me...

>  7z supports swm archives so it should be an easy fix

Famous last words... 7z is only one of the methods Rufus uses to extract from a WIM, when available, so it's not exactly as simple as you may think. But I'm still hoping what you request can be achieved without having to spend a lot of time...

Fuj-, 2015-02-05:
> You do know that you aren't really constrained to use FAT32, and can use NTFS even for UEFI boot. As a matter of fact, the upcoming version of Rufus (Rufus 2.0, an ALPHA of which can be downloaded [...]

I've been wondering why Rufus allows UEFI+NTFS combo. I know UEFI requires FAT so now i'm confused. I don't have access to EFI hw right now so I can't test, but I get no hidden ESP partition on usb, so that's not the workaround. If Rufus is using some non-standard EFI implementation, is it safe, i.e. will it boot on all EFI mobos? And is it present in v1.4 or not (because you say I'm not constrained to FAT32, as in right now, but then you say v2 will seamlessly boot from NTFS on UEFI)?

>7z is only one of the methods Rufus uses to extract from a WIM

I think WIM format had multipart support from the beginning so anything that can handle wim will likely be able to handle swm as well. 7z, wimlib and wimgapi support swm.

Fuj-, 2015-02-05:
>I wasn't aware of this type of split archive beforehand, and find it quite silly that Microsoft went with install.swm, install2.swm, Install3.swm and so on, rather than install.wim, install2.swm, Install3.swm...

Agreed. I think this was a rather obscure method that became of larger interest only recently when install.wim commonly started exceeding 4GB in slipstreamed x64 isos. Interestingly, MS deprecated this in Win8; it can only be installed from a single wim. So if Rufus does the UEFI on NTFS magic, it's extra great news.

pbatard, 2015-02-05:
> I've been wondering why Rufus allows UEFI+NTFS combo.

Read it all in #228.

> I know UEFI requires FAT so now i'm confused.

The thing is: it really doesn't, and some UEFI firmwares do embed an NTFS driver, allowing them to boot NTFS just as well as FAT (in the same way as a Mac EFI firmware will boot HFS natively).

> If Rufus is using some non-standard EFI implementation, is it safe, i.e. will it boot on all EFI mobos?

As far as I know, Rufus follows the UEFI specs: The NTFS driver is UEFI compliant, and it is installed by running a bootloader from an EFI System Partition (which is a 128KB partition placed at the end of the disk, in compliance with the UEFI specs, that state that the ESP can reside anywhere on the disk and doesn't need to be the first).
Now it is possible that some broken UEFI firmwares want the ESP as the first partition, but if they do, they aren't compliant with the specs.

If you want to find out more, please see my [UEFI:TOGO](https://github.com/pbatard/uefi-togo) project (which is a bit of a misnomer, as it is not aimed at Windows To Go only).

> And is it present in v1.4 or not

1.4 provides the ability to create an NTFS partition for an UEFI target, but it does not provide the driver or bootloader to seamlessly boot, so you have to manually go through [this](http://forums.mydigitallife.info/threads/46981-Install-Win8-Windows-8-1-UEFI-With-a-Over-sized%28over-4GB%29-install-wim).

2.0 just completes the picture by removing the extra manual process.

So even with 1.4.12, you aren't actually constrained to FAT32. All you need is an UEFI shell and an NTFS driver to load.

Fuj-, 2015-02-06:
In other words, now it's possible but not entirely practical. :) Looking forward to v2. Thanks for the informative reply and kudos for your work, Rufus is a great piece of software.

Fuj-, 2015-02-09:
Workaround I use for install.swm+uefi scenario: along with .swm I include a minimal install.wim containing only bootmgfw.efi; after Rufus is done I delete install.wim from usb.

commit e4e9e877 (2015-02-09): [misc] handle multipart install.swm Windows images

* Closes #437
closed 2015-02-09 by pbatard