pbatard/rufus #438: Windows 10 error 0x80070570

DasDani, 2015-02-06:
При установке Windows 10, идет установка и где то в 10% выходит ошибка 0x80070570 
"Windows не удается установить необходимые файлы. Возможно файл поврежден или отсутствует"
Пробовал на разных флешках (А-Дата, PQI 16Гб)
Пробовал разные варианта Rufus  и Rufus - The Reliable USB Formatting Utility
Версия 1.4.12 (Build 535)

pbatard, 2015-02-06:
1. I don't speak Russian.
2. I did test Windows 10 Preview 2 installation with Rufus 1.4.12, and found no issue. But you can try the latest [ALPHA](https://rufus.akeo.ie/test/) to see if it improves things. Just note that Windows To Go for Windows 10 (in case you have the option) is not supported. As far as I know, plenty of people have managed to install Windows 10 using Rufus 1.4.12.
3. PLEASE provide a log from Rufus. Without the log, I have no means of finding out your issue.
4. My understanding (from Google translate) is that your issue is not during the creation of the Windows 10 bootable USB, but during the installation onto a computer. In that case, this looks like a pure Windows installation issue, and you may have to look for help elsewhere. Have you tried installing on a hard drive that had been completely erased and without any partitions? Have you tried installing on a different computer?

DasDani, 2015-02-07:
good day.
thank you for answer.
1.I have latest image "Windows 10 Enterprise Technical Preview RUS 9926"
Notebook Lenovo Z570 1Tb. 8Gb DDR3. (Memtest HDD Test is good)
Flash A-Data 8Gb, PQI 16Gb

2. "My understanding (from Google translate) is that your issue is not during the creation of the Windows 10 bootable USB, but during the installation onto a computer."
Yes, problem with instalation. 
When is going installation in 10% error code "Windows 10 error 0x80070570 Perhaps the file is corrupted or missing"  I checked HDD Memory all is good

pbatard, 2015-02-07:
I would still like to see a full log from Rufus. But I don't believe your issue has anything to do with how Rufus copied the files, especially as nobody else reported that error (and I know Rufus has been used by a lot of people to install Windows 10 preview, including myself). What happens if you try with [Microsoft official tool](https://www.microsoft.com/en-us/download/windows-usb-dvd-download-tool). Do you get the same error?

pbatard, 2015-02-14:
No update from OP - closing.

closed 2015-02-14 by pbatard