pbatard/rufus #441: failed to install in mac min(2013)

pianoboysai, 2015-02-10:
test:
ubuntu 14.04 or 14.10 , could be installed, but after installation and reboot,  I always get a grey screen.  the problem was  mentioned in this link
search "THE MANUAL BOOT" 
http://heeris.id.au/2014/ubuntu-plus-mac-pure-efi-boot/

I also tried centos7, failed to install.

pbatard, 2015-02-10:
I read through the guide, and it doesn't look like something I can or even __should__ fix in Rufus.

If I understand properly:
* The creation of the Ubuntu USB media from Rufus is OK
* The boot of the USB media into the Ubuntu installer is OK
* The installation of Ubuntu from the USB media is OK
* The problem is that, __after__ the Ubuntu installer has completed its task, the `grub.cfg` it generated needs to be fixed.

If that is the case, this is a post install task that a tool designed to convert a bootable ISO into a bootable USB should have nothing to do with it. It'd be a bit like asking Rufus to allow you to specify the main account name or serial number to use for a Windows installation: something that is better left to the installer itself.

If the Ubuntu installer is not compatible with the Mac Mini, because it creates a `grub.cfg` that needs to be amended, then you really should ask the Ubuntu people to fix that, not me.

As for CentOS, do you have the same issue on a regular (non Mac) PC?
Again, it's not Rufus job to fix the installer process of standard Linux distributions, if they were simply never designed for Mac.

pianoboysai, 2015-02-11:
ok, thanks for the answer, you are right, it looks like it's the  installer problem.

closed 2015-02-11 by pbatard