pbatard/rufus #442: 2 new USB drives bricked thanks to Rufus? Or just bad drives Help please

almo3361, 2015-02-12:
Hi! Hope someone can help me, as far as I know, I've bricked my ~150$ USB drive. I'm on a win7 computer just so that's clear.
Okay, so I bought a Kingston Datatraveler HyperX 3.0 128GB just a week or so ago. I've formatted it a few times and trying to get it to run a live windows 7 wherever i might be. Didn't work, got bluescreens telling me to run "chkdsk /f". The same problems with another 32GB drive of mine. I didn't bother, I decided I'd rather go with a linux distro, and thought it'd be fun to try something other than ubuntu. I went for Zorin. Downloaded the official ISO and Rufus. I try installing Zoring on the 32GB usb first.

In Rufus, device name was correct for the drive I wanted to use. MBR partition scheme for bios or uefi computers. Large FAT32. 32Kb. Label: Zorin. 
Quick format. Create bootable disk using iso. Create extended label and icon files.

I clicked start, it started and seemed to be working, but after some time I got an error message. I can't remember if it's the same one as I get now ("Error: The device handle is invalid.") or "the system cannot find the file specified usb drive". I'm not 100% sure it was one of those, as I was exhausted while doing this (still am), but I got some error message. The drive was useless after that, I'll go into details afterwards. Since the USB was useless, and I read up on something the creator of Rufus had written, where he said that Rufus can't damage a drive. Okay, decided to test with my new flash drive. Same thing happened with that one too.

Details:
After I got the error message, the USB no longer showed up in the explorer. It is still in computer manager (as RAW), but the options "open", "explore" and "mark as active" are grayed out. "Change drive letter.." and any type of formatting gives the same error message, "The system cannot find the file specified". Clicking on properties literally does nothing.

I ran CHKDSK /F on the drive, it said everything was fine, but did nothing.
I tried formatting it, cleaning it, etc...etc.. in CMD (as admin).
Originally when cleaning I just got some error, but now it manages to clean it, but it still can't format it. Getting the same error message, "The system cannot find the file specified".
In diskpart> list volume the volume is marked as "Unusable". 0B size.
All of this goes for both drives.

Help would be greatly appreciated.

pbatard, 2015-02-12:
Have you tried connecting your drives to a different port, and made sure that you weren't using an extension cable? Especially with USB 3.0 drives, adding an extension cable can produce read or write errors. Also what happens if you plug your drive on an USB 2.0 port? Or on a different computer?

Now, if you tried the above, the fact that you got bluescreens in your live Windows tends to indicate that the problem is with your hardware. As I went great length to explain [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive), it's not because Rufus is repartitioning or reformatting a drive that it is actually doing anything special. As far as the drive is concerned, it's exactly the same operation as writing a file from Windows explorer: some blocks of data are erased and then written, and these blocks are decided by the controller inside the drive - Rufus cannot influence that.

Also, I must point out that Rufus gets downloaded more than 1 million times each month, so if it was damaging drives, it would be widely reported. Instead, what I am seeing is pretty much the number of reports I'd expect from drives that __coincidentally__ happen to fail while someone is using Rufus, which is in the order of at most 2 reports/month at worst (so 2 in a million). And most of these reports end up with _"I have now tried Rufus with another drive (or another computer), and everything was fine"_.

So all I can say, besides the initial things you can try above, is that, __yes__, I am 100% confident that Rufus will not, under any circumstances, incur any more damage to a drive that what would happen if you were simply copying a set of files from Windows explorer, and that bad hardware and coincidences do happen. But these elements have nothing to do with Rufus.

almo3361, 2015-02-12:
That was a fast reply! 
Okay, if that how the software operates (and you should know as you're the one developing it), it can't have been Rufus.

I have tried several USB ports, it was originally in a hub, but I also tried a USB 3.0 port. I didn't try any USB 2.0 port, because I'm lazy and didn't have any unused ones ;)

What perplexes me is that now my new USB drive works again, after going through and experimenting with commands on this page:
http://superuser.com/questions/338059/recovering-a-partially-formatted-usb-thumb-drive

I think what helped was clean, convert it to MBR and then back to GPT, followed by another reboot.
I will try again, but this time with Ubuntu, and see if it ends the same way. if it does, I will return the flash drive.

pbatard, 2015-02-21:
Closing this issue.

closed 2015-02-21 by pbatard