pbatard/rufus #444: USB not connected to Bluray player even though it is

Unitas1, 2015-02-13:
Hi, thank you for your useful utility. I wasn't able to do what I set out to do with it, but that wasn't the fault of Rufus. 

My problem is that once I formatted my USB thumb drive to remove the bootable version of Windows XP I had installed, I could not get my Samsung BD-D5500 Bluray player to recognize the USB.  Oddly enough, the USB thumb drive works fine in my laptop. The disk management tools all say the hard drive is in good shape.  I've formatted the USB about 6 times now, using the quick and longer methods. I've formatted it back to NTFS and FAT32. Neither led my Bluray player to recognize the disc.

Oddly enough, I have another USB thumb drive that I had put Rufus on to create a bootable disc, and my Bluray player does recognize it after removing the OS I had used Rufus to intall.

Could there by some sort of hidden file leftover on my harddrive that's preventing the Bluray player from recognizing the USB? Any suggestions  would be appreciated.

wstrohm, 2015-02-13:
Mr. Batard,

For some reason, ever since I sent you a message regarding a particular problem with Rufus, I have been receiving e-mail messages both to you and from you to different correspondents.

I would appreciate it if you would delete my e-mail address from your distribution.

Thanks for your support.

——    Bill Strohm
wstrohm@socal.rr.com <mailto:wstrohm@socal.rr.com>wstrohm@socal.rr <mailto:wstrohm@socal.rr>.com

pbatard, 2015-02-13:
@wstrohm (Bill Strohm)

It looks like __you__ added yourself to the [list of people who watch Rufus](https://github.com/pbatard/rufus/watchers), in which case you will receive an e-mail any time an issue is created or updated.

I have __no__ control over this list, and especially, even as I own the project, I cannot remove anyone from it.

If you want to stop receiving these e-mail, you will have to go to https://github.com/pbatard/rufus and click the `Unwatch` button.

@Unitas1,

Are you sure you didn't format your drive in GPT mode instead of MBR, or that your drive isn't using 4K sectors (instead of 512 bytes).
If you could provide a __full log__ from Rufus, it will give me more details on how you formatted your drive, and I might be able to help you. Otherwise, I really have no idea what the problem can be. It is also very possible that your Samsung player has limitations with regards to some drive, in which case there is not much Rufus will be able to do.

Unitas1, 2015-02-13:
Hi, thanks. The Samsung player previously played this particular USB thumb drive with no problems.  The problem started immediately post-Rufus.  Here is the log.

Rufus version: 1.4.12.535
Syslinux versions: 4.07/2013-07-25, 5.10/2013-06-04
Windows version: Windows 7 SP1 64-bit
Locale ID: 0x0409
Found USB 2.0 device 'STT EXPRESS ST1 USB Device' (1B8F:0311)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 938, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x068CB211
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.2 GB (7720665088 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Unitas1, 2015-02-13:
Ok. I formatted the USB with Rufus and now the  Samsung player recognizes the USB thumb drive again.  Previously I had used Windows to format the drive.  Thanks for putting me on the right track, and thanks for the great utility.

pbatard, 2015-02-14:
OK, I'm going to close this issue then.

closed 2015-02-14 by pbatard