pbatard/rufus #445: device was eliminated because it didn't report itself as disk 1.4.12.535

milesj098, 2015-02-14:
[image]

pbatard, 2015-02-14:
1. Your image doesn't display
2. __PLEASE__ copy/paste a full log! There's __NOTHING__ I can do unless you provide a log.
3. If your device is an USB HDD then this is normal. See [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected). And as the log indicates, you can easily get your device listed by going to the advanced mode.
4. If your device is not an USB HDD, then I will definitely need the log, as well as the details of your device (manufacturer and exact model).

milesj098, 2015-02-15:
gimme a sec i tried to put it on but i guess it didnt
im trying to save the log but it wont work it keeps poping up url or somethin

pbatard, 2015-02-15:
Just copy/paste the content of the log in your reply. Also please indicate if your device is an external HDD or something else.

pbatard, 2015-02-21:
No update from OP &rarr; Closing.

closed 2015-02-21 by pbatard