pbatard/rufus #446: "Error reading UDF file \boot\en-us\bootsect.exe.mui" multiple times on different ISO files

BegbertBiggs, 2015-02-19:
Now, I realize that there were previous reports of this problem already. When I first had the issue, I looked at them and came to the conclusion that the file was corrupted. No biggie, I'll get a new one. But as soon as I try the other one, I get the same issue. So my question is: *Can there really be no other reason for the problem than a corrupted file?* It just seems weird to me that it happened with completely different files.

Full log:

    Rufus version: 1.4.12.535
    Syslinux versions: 4.07/2013-07-25, 5.10/2013-06-04
    Windows version: Windows 7 SP1 32 bit
    Locale ID: 0x0407
    Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C753)
    Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
    Device eliminated because it appears to contain no media
    1 device found
    Disk type: Removable, Sector Size: 512 bytes
    Cylinders: 974, TracksPerCylinder: 255, SectorsPerTrack: 63
    Partition type: MBR, NB Partitions: 1
    Disk ID: 0x0EE12308
    Drive has a Rufus Master Boot Record
    Partition 1:
      Type: NTFS (0x07)
      Size: 7.5 GB (8018460672 bytes)
      Start Sector: 2048, Boot: Yes, Recognized: Yes
    Scanning image...
    Disc image is an UDF image
    ISO label: 'IR1_CPRA_X64FREV_EN-US_DV5'
      Size: 3928639488 bytes
      Has a >64 chars filename: No
      Has Symlinks: No
      Has a >4GB file: No
      Uses ReactOS: No
      Uses KolibriOS: No
      Uses EFI: Yes
      Uses Bootmgr: Yes
      Uses WinPE: No
      Uses isolinux: No
    Using ISO: en_windows_8_1_pro_vl_x64_dvd_2971948.iso

    Format operation started
    Requesting disk access...
    Opened drive \\.\PHYSICALDRIVE2 for write access
    Will use 'H:' as volume mountpoint
    I/O boundary checks disabled
    Analyzing existing boot records...
    Drive has a Rufus Master Boot Record
    Volume has an unknown Partition Boot Record
    Deleting partitions...
    Clearing MBR/PBR/GPT structures...
    Erasing 128 sectors
    Partitioning (MBR)...
    Closing existing volume...
    Waiting for logical drive to reappear...
    Formatting (NTFS)...
    Using cluster size: 4096 bytes
    Quick format was selected
    Creating file system: Task 1/12 completed
    Creating file system: Task 2/12 completed
    Creating file system: Task 3/12 completed
    Creating file system: Task 4/12 completed
    Creating file system: Task 5/12 completed
    Creating file system: Task 6/12 completed
    Creating file system: Task 7/12 completed
    Creating file system: Task 8/12 completed
    Creating file system: Task 9/12 completed
    Creating file system: Task 10/12 completed
    Creating file system: Task 11/12 completed
    Creating file system: Task 12/12 completed
    Format completed.
    Writing master boot record...
    Drive has a Zeroed Master Boot Record
    Set bootable USB partition as 0x80
    Using Rufus MBR
    Found volume GUID \\?\Volume{bbac017b-7e2f-11e1-9da6-406186f2cf97}\
    Opened drive \\?\Volume{bbac017b-7e2f-11e1-9da6-406186f2cf97} for write access
    Writing partition boot record...
    Using Standard NTFS partition boot record
    Confirmed new volume has an NTFS boot sector
    Successfully remounted Volume{bbac017b-7e2f-11e1-9da6-406186f2cf97}\ on H:\
    Copying ISO files...
    Extracting files...
    Disc image is an UDF image
    Extracting: H:\autorun.inf (128 bytes)
    Extracting: H:\boot\bcd (256 KB)
    Extracting: H:\boot\boot.sdi (3 MB)
    Extracting: H:\boot\bootfix.bin (1 KB)
    Extracting: H:\boot\bootsect.exe (106.8 KB)
    Extracting: H:\boot\en-us\bootsect.exe.mui (15.5 KB)
      Error reading UDF file \boot\en-us\bootsect.exe.mui

    Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C753)
    Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
    Device eliminated because it appears to contain no media
    1 device found
    Disk type: Removable, Sector Size: 512 bytes
    Cylinders: 974, TracksPerCylinder: 255, SectorsPerTrack: 63
    Partition type: MBR, NB Partitions: 1
    Disk ID: 0x0FC67003
    Drive has a Rufus Master Boot Record
    Partition 1:
      Type: NTFS (0x07)
      Size: 7.5 GB (8018460672 bytes)
      Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-02-20:
> When I first had the issue, I looked at them and came to the conclusion that the file was corrupted.

The issue is that some of the ISO images that Microsoft provides appear to be corrupted (yes, __official__ images, though AFAIK this should be limited to some Windows 7 ISOs). This means that you can download the file as many times as you want, it will still be corrupted.

Still, I would __strongly__ encourage you to compute the sha1sum of your `en_windows_8_1_pro_vl_x64_dvd_2971948.iso` and confirm that you get the same value as the one from the MSDN. It should be `CFCDF895679A5116F7F1C6406C20F99A1F850E49`.

If `CFCDF895679A5116F7F1C6406C20F99A1F850E49` is the actual hash you get for your ISO, then I'll see if I can get my hands on it to test. But even then, I'm still leading towards a non UDF compliant ISO produced by Microsoft, since they are already known to have generated problematic ones.

BegbertBiggs, 2015-02-20:
I found another solution, thank you for the reply anyways.

closed 2015-02-20 by BegbertBiggs

pbatard, 2015-02-21:
Note: I have now tested `en_windows_8_1_pro_vl_x64_dvd_2971948.iso` (sha1sum `CFCDF895679A5116F7F1C6406C20F99A1F850E49`) and found no issue, therefore I believe that the problem was entirely on your side.

Please make sure you always use official Microsoft ISO images, and confirm that the ISO hash matches the one from the MSDN.