pbatard/rufus #447: Rufus 1.4.12: Undetermined Error while Formatting

kukubau, 2015-02-21:
Hi

I've been trying to build today a bootable USB flash drive using a Windows 8 Core/Pro iso for reinstalling the OS on my laptop.

I've tried several times using a 32 gb SanDisk Cruzer Fit usb flash drive, but in avail, even though I could very well write the *.iso using UltraIso.
I'm attaching the log file here:

Rufus version: 1.4.12.535
Syslinux versions: 4.07/2013-07-25, 5.10/2013-06-04
Windows version: Windows 8.1 64-bit
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4601DF81-74EE-4D8F-B716-C019D0BC4C5A}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {84643ED4-B4DD-4392-BDA8-F366EAABD7E0}
  Size: 29.8 GB (32014614016 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
Disc image is an UDF image
ISO label: 'IR3_CCSA_X64FRE_EN-US_DV9'
  Size: 4191512576 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses ReactOS: No
  Uses KolibriOS: No
  Uses EFI: Yes
  Uses Bootmgr: Yes
  Uses WinPE: No
  Uses isolinux: No
Using ISO: en_windows_8.1_with_update_x64_dvd_4065090.iso
WIM extraction method(s) supported: wimgapi.dll

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'E:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system: Task 1/5 completed
Creating file system: Task 5/5 completed
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {743B9384-9615-4278-97F4-E3CB1D2B3D03}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {A680A7D3-6CC4-4751-96E0-3C056F54BB25}
  Size: 29.8 GB (32014614016 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {743B9384-9615-4278-97F4-E3CB1D2B3D03}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {A680A7D3-6CC4-4751-96E0-3C056F54BB25}
  Size: 29.8 GB (32014614016 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

And the 2nd log running Rufus after successfully writing the Iso using UltraIso 9.6.2.3059 (I don't know if it was UEFI compatible, but it did write it).

Rufus version: 1.4.12.535
Syslinux versions: 4.07/2013-07-25, 5.10/2013-06-04
Windows version: Windows 8.1 64-bit
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
No volume information for drive 0x82
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xCAD4EBEA
Drive has an unknown Master Boot Record
Partition 4:
  Type: FAT32 LBA (0x0c)
  Size: 29.8 GB (32015548416 bytes)
  Start Sector: 256, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: 'IR3_CCSA_X64FRE_EN-US_DV9'
  Size: 4191512576 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses ReactOS: No
  Uses KolibriOS: No
  Uses EFI: Yes
  Uses Bootmgr: Yes
  Uses WinPE: No
  Uses isolinux: No
Using ISO: en_windows_8.1_with_update_x64_dvd_4065090.iso
WIM extraction method(s) supported: wimgapi.dll

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
No drive letter was assigned...
Will use 'E:' as volume mountpoint
Analyzing existing boot records...
Drive has an unknown Master Boot Record
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 16384 bytes
Quick format was selected
Creating file system: Task 1/5 completed
Creating file system: Task 5/5 completed
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'E:' after error

Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
No volume information for drive 0x82
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4E8DE1BC-8819-426B-9EF8-2BDA10371884}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {9A610FA7-A716-4B4E-8DD2-15EC31F3C3B0}
  Size: 29.8 GB (32014614016 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
1 device found
No volume information for drive 0x82
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4E8DE1BC-8819-426B-9EF8-2BDA10371884}
Max parts: 128, Start Offset: 17408, Usable = 32015645184 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {9A610FA7-A716-4B4E-8DD2-15EC31F3C3B0}
  Size: 29.8 GB (32014614016 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Ty

pbatard, 2015-02-21:
1. What happens if you try to format the drive using the default Windows format tool? Does it work?
2. have you tried the [latest BETA](https://rufus.akeo.ie/downloads/)?
3. What happens if you try to format it as MBR and then GPT? Does MBR formatting work?

kukubau, 2015-02-22:
Hi

I've just tested Rufus on 2 laptops. The 1.4.12 hangs every time with the above mentioned error when trying to format the flash drive as GPT, but only on one laptop (i7-Haswell)(the log files I've attached come from that one). MBR for BIOS and MBR for BIOS and EFI works fine on both.

Rufus 2.0 Beta doesn't throw me any errors. It formats GPT well without issues on both as well as MBR for BIOS and MBR for BIOS and EFI.

1.4.10 works with GPT as it previously did on both, no issues here either. And neither with MBR for BIOS and MBR for BIOS and EFI.

pbatard, 2015-02-23:
> Rufus 2.0 Beta doesn't throw me any errors. It formats GPT well without issues on both as well as MBR for BIOS and MBR for BIOS and EFI.

OK. Then I'm going to close this issue, as 2.0 is going to be released soon.

closed 2015-02-23 by pbatard