pbatard/rufus #450: Windows To Go 8.1 "ISO Extraction Failure"

cookieboyeli, 2015-02-25:
I tried to create a Windows To Go installation on my 32GB Sandisk Extreme using a custom Windows ISO (this could be the problem). It has all updates until February 2015 included in the installation which saves years of my life :). It's using a boot.wim not boot.esd so that shouldn't be the issue.

```Rufus version: 2.0.632 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 1.0 device 'Myson CS8819A3-116  0 USB Device' (04CF:9920)
Device eliminated because it appears to contain no media
Found USB device 'Myson CS8819A3-116  1 USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB device 'Myson CS8819A3-116  2 USB Device' (????:????)
Found USB device 'Myson CS8819A3-116  3 USB Device' (????:????)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8005877760 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB 3.0 device 'Myson CS8819A3-116  0 USB Device' (0781:5580)
Device eliminated because it appears to contain no media
Found USB device 'Myson CS8819A3-116  1 USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB device 'Myson CS8819A3-116  2 USB Device' (????:????)
Found USB device 'Myson CS8819A3-116  3 USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 1.0 device 'SanDisk Extreme USB Device' (0781:5580)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.5 GB (8005877760 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB 1.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x04513346
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.2 GB (31375658496 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: Win81-x64-Feb2015
  Size: 6461894656 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win81-x64-en-US-Feb2015.iso

[... 1165 lines cut ...]
No further action is required.

  30640291 KB total disk space.
   6343836 KB in 936 files.
       288 KB in 102 indexes.
     44519 KB in use by the system.
     41888 KB occupied by the log file.
  24251648 KB available on disk.
      4096 bytes in each allocation unit.
   7660072 total allocation units on disk.
   6062912 allocation units available on disk.
NTFS Fixup completed.

Found USB 1.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08E55669
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.2 GB (31375658496 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

pbatard, 2015-02-25:
Windows To Go requires a `sources\install.wim` Windows Image file, with the To Go files at index nr. 1.
Do you have that on your ISO?

The following from the log seems to indicate that you are missing the file:
```
Opening: \\.\CDROM0\sources\install.wim:[1]
  Could not access image: [0x00000002] The system cannot find the file specified.
Failed to apply Windows To Go image
```

For the record, I have no plans to support non official To Go images.
If you want the Windows To Go feature of Rufus to work with your image, you must to make sure that your ISO is created in the same manner as official Microsoft ones.

cookieboyeli, 2015-02-25:
I just looked for /sources/install.wim and it doesn't exist. there is however an equivalently large file where it would be named install.swm and install.swm2. It looks like install.swm is just a hair under the 4GB limit which is probably why this split swm method was used.

I'm guessing adding support for this would be quite time consuming?

pbatard, 2015-02-25:
#437 should already have added support for `install.swm`. I tried to test it at the time, and thought it worked, but maybe I overlooked something (and the person who requested it didn't get back to me indicating that there was an issue).

Note however that you do __NOT__ need to split your `install.wim` when using Rufus even if it's larger than 4GB, as Rufus enable __seamless__ NTFS boot from UEFI (and hence support for >4GB UEFI files) through [UEFI:TOGO](https://github.com/pbatard/uefi-togo). But even the 4GB issue (or lack thereof) should only apply for a Windows standard install, and not Windows To Go anyway, since To Go will apply (extract) the files from `install.wim` instead of copying it wholesale.

I'll see if I can re-test `install.swm` support. I think the original issue was with extracting the Windows 7 EFI bootloader rather than To Go, so that may be why it doesn't work.

cookieboyeli, 2015-02-27:
I think I should note that it's not my ISO. It's meant as an all purpose, fully updated, x64 Windows installation ISO. I use it for installing on BIOS machines and UEFI machines.
Splitting may not be beneficiary in this case, but if it can be split without issues here, then that just increases compatibility and overall configurations it can be used in.

pbatard, 2015-02-27:
I expect [BETA 4](https://rufus.akeo.ie/downloads/) to fix your issue. Please test and feel free to reopen it if it doesn't work.

closed 2015-02-27 by pbatard

cookieboyeli, 2015-02-28:
Ok it's written without error, however I'm having trouble even detecting it in the bios. Should I create a new issue for this? (I'm 99% certain that my MB bios settings are correct).

Just for good measure here is the log.
```
Rufus version: 2.0.634 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x08E55669
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.2 GB (31375658496 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: Win81-x64-Feb2015
  Size: 6461894656 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win81-x64-en-US-Feb2015.iso
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding MSR partition
Adding extra partition
Reserved 3251 tracks (100 MB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Closed Windows format prompt
Using cluster size: 4096 bytes
Quick format was selected
Creating file system: Task 1/12 completed
Creating file system: Task 2/12 completed
Creating file system: Task 3/12 completed
Creating file system: Task 4/12 completed
Creating file system: Task 5/12 completed
Creating file system: Task 6/12 completed
Creating file system: Task 7/12 completed
Creating file system: Task 8/12 completed
Creating file system: Task 9/12 completed
Creating file system: Task 10/12 completed
Creating file system: Task 11/12 completed
Creating file system: Task 12/12 completed
Format completed.
Found volume GUID \\?\Volume{a3627018-08bc-4e14-a6d5-fdc303e26590}\
Volume is already mounted, but as D: instead of G: - Unmounting...
Successfully remounted Volume{a3627018-08bc-4e14-a6d5-fdc303e26590}\ on G:\
Applying Windows image...
Windows To Go mode selected
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Mounted ISO as '\\.\CDROM0'
Opening: \\.\CDROM0\sources\install.swm:[1]
Applying Windows image...
Closing: \\.\CDROM0\sources\install.swm
Setting up MS EFI system partition
Successfully mounted '\Device\HarddiskVolume10' (USB partition 3) as 'D:'
Formatting EFI system partition D:
Creating file system: Task 1/5 completed
Creating file system: Task 2/5 completed
Creating file system: Task 3/5 completed
Creating file system: Task 4/5 completed
Creating file system: Task 5/5 completed
Enabling boot: 'G:\Windows\System32\bcdboot.exe G:\Windows /f ALL /s D:'
Boot files successfully created.
Successfully unmounted 'D:'
Applying san_policy.xml...

Deployment Image Servicing and Management tool
Version: 6.3.9600.17031
Image Version: 6.3.9600.17031
The operation completed successfully.
Copying 'unattend.xml'
Finalizing, please wait...
Created: G:autorun.inf
Created: G:autorun.ico
NTFS Fixup (Checkdisk)...
Volume label is Win81-x64-Feb2015.

Stage 1: Examining basic file system structure ...
  108800 file records processed.
File verification completed.
  744 large file records processed.
  0 bad file records processed.

Stage 2: Examining file name linkage ...
  151942 index entries processed.
Index verification completed.

Stage 3: Examining security descriptors ...
Security descriptor verification completed.
  21571 data files processed.

Windows has scanned the file system and found no problems.
No further action is required.

  30406792 KB total disk space.
   9177220 KB in 86291 files.
     67828 KB in 21573 indexes.
    151916 KB in use by the system.
     41600 KB occupied by the log file.
  21009828 KB available on disk.
      4096 bytes in each allocation unit.
   7601698 total allocation units on disk.
   5252457 allocation units available on disk.
NTFS Fixup completed.

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {E7D2DD8E-909C-42DB-AC8A-75B1295C0207}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {C2FA5FAD-8215-4943-A32B-695EC5FD4EC8}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {A3627018-08BC-4E14-A6D5-FDC303E26590}
  Size: 29.0 GB (31136555520 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {5775B9C5-EE93-49B5-997D-8AABC12B608F}
  Size: 100 MB (104864256 bytes)
  Start Sector: 61077777, Attributes: 0x0000000000000000
```

pbatard, 2015-02-28:
> when it actually loads I get "windows installation error occurred and must restart" or something like that.

Please try with an official Microsoft ISO then (other than Windows 10, as I found Windows 10 To Go didn't work too well at least for the x64 version, and it's still BETA, so there's no guarantee that Windows 10 To Go was properly finalized). There's very little that Rufus has an influence on after boot, so if it boots, and you get an error further down the process on unofficial ISOs, you will have to sort it out yourself.

For what is worth, in GPT/UEFI mode, Rufus creates a Windows To Go drive that is as close as possible to what the offiicial Microsoft recommends (with MSR and UEFI System Partitions), so I doubt the issue is with Rufus...

> Or perhaps there's a magic code to speed it up?

No there isn't.

Extraction for Windows installation uses Rufus internal code (through the Open Source library libcdio), so it's fast. 

Extraction (`apply-image`) for Windows To Go uses Microsoft APIs, because there's no documentation, that I could find, on what `apply-image` really does (it's not as simple as simply extracting the WIM data), and there doesn't exist an Open Source library to do the same.

That's part of the reason why Windows To Go is not supported in Rufus for Windows 7, as the Microsoft APIs require the `.wim` to be accessible as a regular file, and therefore we need the ability to natively mount ISOs, which only Windows 8 provides.

Unfortunately, it doesn't look like Microsoft spent a lot of time trying to optimize their WIM image application code, in fact, even their progress estimate on image application is so screwed up that I had to compensate for it in Rufus, so WIM image extraction is __very, very__ slow. I'm afraid this is not something I'm gonna be able to change unless I were to invest something like 3 __full__ months to create an Open Source WIM image application library, which I'm not gonna do.

cookieboyeli, 2015-02-28:
I guess I expected that. 7 minutes is not that long anyway. When USB 3.1 gets going that'll probably be 4 minutes. (USB 3.1 Type C FTW!)

I'll try with the official ISO and see if that helps. 

I guess this issue is sorted out now! Thanks! :)

pbatard, 2015-03-01:
OK.

For the record I have just been made aware that there exists an Open Source WIM library, so I'll try to see if I can use that in a future version of Rufus (#453).

cookieboyeli, 2015-03-01:
Oh neat! I'll be looking forward to it! :)

Fuj-, 2015-03-05:
> #437 should already have added support for install.swm. I tried to test it at the time, and thought it worked, but maybe I overlooked something (and the person who requested it didn't get back to me indicating that there was an issue)

There is no issue, apart from the documented fact that installing from .swm is supported on Win7 only. I made a mention of it in #437:

> Interestingly, MS deprecated this in Win8; it can only be installed from a single wim.

pbatard, 2015-03-05:
Yup, I realize as I was fixing the issue reported by @cookieboyeli that the only thing I had done until then for `.swm` support had to do with Windows 7 installation, and not Windows to Go.

The current version should support `.swm` for both Windows 7 installation and Windows To Go.

commit 000f60c5 (2015-02-27): [core] improve support for multipart install.wim

* Closes #450