pbatard/rufus #452: Windows To Go 8.1+ on removable disks

FrozenCow, 2015-02-28:
It would be great if Windows To Go works on removable disks as well. I'm not sure what is needed for this, but I have had it working for Windows 8 using [this method](http://softwarebakery.com/boot-windows-from-drivedroid). Somehow Windows 8.1 and 10 do not have support for removable disks.

A bit of background: I'm working on [an application for Android](https://play.google.com/store/apps/details?id=com.softwarebakery.drivedroid) that allows people to use their phone as an USB drive or CD-rom drive. The functionality to act like such a drive is baked into the Linux kernel of most phones. The functionality has support for enabling/disabling the 'removable' flag, but phones are not shipped with this. People can patch their kernels and flash it on their phones to enable/disable this flag, but it is a huge pain to do so and there a not many people willing to go through all this.

This is why I'm interested whether it can be done from Windows. From what I can see in the code of the kernel that emulates the USB drive, there are no extra SCSI commands for 'fixed' disks. So technically it should be possible. The fact that Windows 8 also did work proves it is possible. I'm thinking it might be an artificial limitation in Windows 8.1+ where it just checks for the 'fixed/removable' flag on the device to determine whether it is a bootable device.

I don't have much knowledge of the internal workings of Windows nor its boot process, so I was wondering whether you found anything that could remove the limitation and allows 'to go' on removable drives?

pbatard, 2015-03-01:
OK, first of all, Rufus should only prevent the creation of Windows To Go on removable media if you are using GPT/UEFI. If you use MBR/UEFI, it should let you use removable (albeit with a warning that this may not work, as I have never had any success using Windows 8.1, but some people have stated that Windows 8.0 and 7 To Go might work on removable). The reason this is prevented altogether for GPT/UEFI is that I found that Windows To Go would simply not boot without an EFI System Partition and an MSR, and creating those on a removable drive becomes very problematic (since Windows will only ever be able to mount the first partition with a filesystem it recognizes, and there's nothing you can do to get Windows to see the other ones at the same time).

Second, AFAIK, the only way to enable removable support for To Go in 8.1 or later would be to patch the Windows kernel. Unfortunately, this is a __massive__ undertaking, since the Windows kernel is closed source and the debug info (which I tried to check), gives no hint whatsoever about why exactly Windows refuses to boot. Everyone's guess of course, including mine, is that it checks for the REMOVABLE vs FIXED flag, but that remains just a guess. I actually [asked about this on reboot.pro](http://reboot.pro/topic/20280-windows-to-go-81-on-‘removable’-easy2boot-drive-both-mbr-‘’-uefi-‘possible’/?p=190420), but it doesn't look like anybody wants to take on that challenge:

> Since this is a forum dedicated to figuring out this kind of things, wouldn't it be nice if someone could take a look at the Windows boot process and try to come up with a patch to make 8.1 W2G also boot on removable?...

I sure wouldn't mind having all the freedom and time in the world to look at the Windows kernel, to try to locate where the FIXED/REMOVABLE check might occur and patch it (because I actually LOVE analysing disassembly, and that sounds like a nice challenge). However it's hard to justify this kind of investment unless you are paid to do so as part of your day job, since reverse engineering can and usually is __extremely__ time consuming... Sadly, while the world needs MORE independent reverse engineers, there doesn't seem to be any company out there willing to foot that kind of bill... :disappointed:

FrozenCow, 2015-03-01:
Yes, I was afraid it would be quite a time investment to figure things out. I hoped it might've been some .sys file that was responsible for this (which might allowed for an easy Windows 8 with Windows 8.1 .sys file swap), but since this is related to booting the filesystem it is indeed probably in the kernel.

Thanks for the thorough response though. The link explains quite a bit. Hopefully things change a bit with Windows 10 in this regard, probably not, but one can only hope ;)

closed 2015-03-01 by FrozenCow