pbatard/rufus #455: Please add "No update" confirmation dialog.

cookieboyeli, 2015-03-03:
It is currently unclear if Rufus is connecting to the update server and actually checking or not as there is no "No update" confirmation with there isn't a new release/beta.

Also, if auto check for updates is disabled, check for beta edition is greyed out. Does this mean when you manually check it does NOT check for beta editions? It's impossible to definitively tell.

pbatard, 2015-03-03:
I disagree. If you don't get a popup, and have check for update enabled, then there wasn't any update. And if that's not enough, you can always issue a manual check. And if that still isn't enough, the __log__ is here for you!

Also, if you're relying on an UI to tell you black on white what an application does, you're going to have a bad time. That's because not matter how carefully you try to craft an UI, one person will always have a different opinion of what an UI state means, and there is no such thing a "common agreed sense"here. On the other hand, there is such thing as trying to interpret what an UI means, __and__ failing to interpret what its author tried to convey...

If you really want to find out what happens during the update check, and whether the BETA will be queried, just try to change the beta option and look at the log. Rufus will tell you when/if it's checking the _Release_ and _Beta_ channels there.

Closed as "not gonna happen"... ;)

cookieboyeli, 2015-03-03:
Ok, I was talking about the manual check though. It seemed weird to me as I clicked "check for update" and nothing happened. I figured a "less advanced" user would not find that satisfying enough, and that it would be simple to use the same window that says "you have an update" to say "you don't have na update" 

But now that I think about it more, there aren't really "less advanced" users like I'm thinking of using this. It requires a certain amount of technical proficiency to understand.

So good call. :)

cookieboyeli, 2015-03-06:
Hold on a second! I just realized the REAL problem. I didn't even see the "Checking for Rufus updates..." status message at all since it was covered by the about window. That's why I thought there was no notification.

A simple tweak would be to raise the default positioning of the about window to meet with the top of the main window. This is fine since the main window is unclickable and you see the version number in the about window. It also lines up exactly to give the status bar/box exactly the right margin.

I know it's extremely minor bordering on inconsequential but it's like a 5 minute tweak tops and it makes update status visible when clicking update instead of just a "dead button" feeling.

pbatard, 2015-03-06:
> but it's like a 5 minute tweak tops

Seeing how long I spent making sure that the log window appeared where I wanted on the side of the main application, I can tell you it's not. Please see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

Also this is only an issue if you __manually__ check for updates (in which case, if you don't see anything, it means that there wasn't any update). For the regular background update check, the "No update" message will be visible in the application.

I may review this when I have nothing better to do, but since this isn't going to be the 5 minute tweak you think it is, you might be in for a long wait.

cookieboyeli, 2015-03-06:
Ah yes, I read that when I first found Rufus. I should have known!

Thanks for considering pbatard.

pbatard, 2015-03-06:
Now that I had more time to think about it, I'll see if I can squeeze in your initial suggestion of issuing a dialog notification in case there is no update, as this is the method that should require the least amount of work.

cookieboyeli, 2015-03-06:
Ok thanks. :)

commit 0f4067f5 (2015-03-09): [misc] add notification on forced update check when no update is found

* Closes #455
closed 2015-03-09 by pbatard