pbatard/rufus #458: Empty files form iso missing

niteflyer, 2015-03-09:
A diff between the ISO and the filesystem on the USB stick showed, that Rufus had replaced/removed some files, which might (or might not?) be ok in case of driver files (Windows XP install disk, Rufus 2.0 default settings, with disabled Autorun.inf handling): anyway, XP setup runs fine. 

But recently I have noticed, that Rufus also has dropped an empty file  in \$OEM$\\$1\INSTALL. Is this a bug or side effect of disk optimization? I don' think it's ok, to break the ISO in that way:

I had put this file there (NODPACKS.TXT) for debugging purposes to let PRESETUP.CMD decide whether or not to extract its DriverPacks to the SystemDrive (if exist NODPACKS.TXT exit), so that I don't need to rebuild the ISO for this purpose... Obviously, this test fails, if Rufus removes the file. Currently, I'm helping myself and put some text into the file.

pbatard, 2015-03-09:
As __always__, it would really help to get a full log from your USB creation in Rufus, because it would answer the following:
* What version of Rufus are you using? I think some older versions use to have an issue with empty files, but that was fixed
* What was your target filesystem?
* What OS are you running Rufus on?

I also wouldn't mind knowing how you mastered your XP image. What application did you use to create the ISO?

For what is worth, I have re-tested that Rufus 2.0 does properly handle an empty `$OEM$\$1\INSTALL`. I used a Syslinux based boot image, that I created with ImgBurn. For good measure, I also added an zero byte `empty.txt` at the root level. See log below:
```
Extracting files...
libcdio: Found Extension: Joliet Level 3
Disc image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: F:\$OEM$\$1\INSTALL. (0 bytes)
Extracting: F:\empty.txt (0 bytes)
Extracting: F:\live\filesystem.packages (29.2 KB)
Extracting: F:\live\initrd.img (8.6 MB)
Extracting: F:\live\vmlinuz. (2.3 MB)
Extracting: F:\syslinux\boot.cat (2 KB)
Extracting: F:\syslinux\exithelp.cfg (63 bytes)
Extracting: F:\syslinux\f1.txt (842 bytes)
Extracting: F:\syslinux\f10.txt (728 bytes)
Extracting: F:\syslinux\f2.txt (424 bytes)
Extracting: F:\syslinux\f3.txt (597 bytes)
Extracting: F:\syslinux\f4.txt (357 bytes)
Extracting: F:\syslinux\f5.txt (807 bytes)
Extracting: F:\syslinux\f6.txt (1.1 KB)
Extracting: F:\syslinux\f7.txt (933 bytes)
Extracting: F:\syslinux\f8.txt (181 bytes)
Extracting: F:\syslinux\f9.txt (747 bytes)
Extracting: F:\syslinux\isolinux.bin (24 KB)
Extracting: F:\syslinux\isolinux.cfg (68 bytes)
Extracting: F:\syslinux\live.cfg (2.8 KB)
Extracting: F:\syslinux\menu.cfg (88 bytes)
Extracting: F:\syslinux\prompt.cfg (174 bytes)
Extracting: F:\syslinux\splash.png (16.8 KB)
Extracting: F:\syslinux\stdmenu.cfg (518 bytes)
Extracting: F:\syslinux\syslinux.cfg (67 bytes)
Extracting: F:\syslinux\vesamenu.c32 (159.3 KB)
Created: F:\syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico
```
I confirmed that both the zero byte `empty.txt` and `$OEM$\$1\INSTALL` were created as expected. I also tried both NTFS and FAT32 as target file systems and found no issue.

Please provide a log and indicate how your ISO was created.

pbatard, 2015-03-15:
No update from OP. Closing.

closed 2015-03-15 by pbatard