pbatard/rufus #459: ER: Access to the device is denied

TNTCookie, 2015-03-10:
Everytime I try to use Rufus on my USB it gives an error saying: "Access to the device is denied"
_______________________________________________________________________________
Rufus version: 2.0.639
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: ESD-ISO
  Size: 3519393792 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: ESD-ISO
  Size: 3519393792 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: ESD-ISO
  Size: 3519393792 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Windows.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7a717372-c9d2-11e3-825c-fcf8ae8c1229}\ was already mounted as H:\
Re-mounted volume as 'H:' after error

Found USB 2.0 device 'Imation ImationFlashDriv USB Device' (0718:067D)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 487, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xC3072E18
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.7 GB (4003135488 bytes)
  Start Sector: 8064, Boot: No, Recognized: Yes

pbatard, 2015-03-10:
You should pay more attention to what the log explicitly tells you:
```
ABORTED: Cannot use an image that is located on the target drive!
```

The `Windows.iso` you are trying to use resides on the same drive as the one you are trying to format. You are trying to saw the branch you're sitting on - what you are trying to do was never going to work in the first place!

Please move your `Windows.iso` to a different drive than your USB, and be mindful that __ALL__ data on your USB drive will be erased

closed 2015-03-10 by pbatard

TNTCookie, 2015-03-10:
Well that was stupid of me.
You thought me a lesson. To read the log.

kourindouhime, 2016-05-14:
I'm glad that issue exists. Thanks.

coder618, 2017-03-25:
I think its happen, when your Selected ISO file location and Selected USB drive is same .

pbatard, 2017-03-25:
@coder618, this is exactly what I explained above. The `ABORTED: Cannot use an image that is located on the target drive!` message from the log is pretty explicit about that.

clintonastute, 2018-11-15:
> You should pay more attention to what the log explicitly tells you:
>
> ```
> ABORTED: Cannot use an image that is located on the target drive!
> ```
> The `Windows.iso` you are trying to use resides on the [...]

Thank you so much. This helped me.

clintonastute, 2018-11-15:
@pbatard Thank you so much. This helped me.

ntnatow, 2018-12-05:
Yeaaaah, I may have made this mistake today as well. Lesson learned.

FahmiBmukbil, 2019-01-16:
Hello!
I get this error even though the ISO is on different drive + I found no errors in the log...
Does it mean it's 100% successful?

pbatard, 2019-01-16:
Don't know __unless__ you attach your log. Also, the __only__ way to tell if the device was created successfully is to boot from it. have you tried that?

But FYI, there are some operations where Rufus will automatically retry if the access is designed. If Rufus says that the operation is complete, then it __is__ complete.

FahmiBmukbil, 2019-01-16:
1. Yes I tried booting up from it, I get this:
[image]

2. Yes it says it's complete!

pbatard, 2019-01-16:
Then your issue has nothing to do with Rufus. All the files were copied, so if it doesn't boot, you'll have to look elsewhere for answers.

FahmiBmukbil, 2019-01-16:
hmmm, good to know!
thx