pbatard/rufus #460: Not Showing Options

Danialde4, 2015-03-11:
Here's the problem:

http://i.imgur.com/B4PoCr2.png

pbatard, 2015-03-11:
http://rufus.akeo.ie/where_is_the_log
Also https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected

Also, that picture doesn't describe anything. This is what you should expect if you don't have a drive inserted or if you have an USB HDD and haven't enabled listing of HDD in advanced options. So it __doesn't__ give much hint about your issue.

Danialde4, 2015-03-11:
Thanks a lot. It really helped me. :+1:

Danialde4, 2015-03-11:
Here's the result 
[image]

pbatard, 2015-03-11:
So I'm guessing it solved your problem?
Can you confirm if your USB device an USB hard drive?
If it's not I will need your log so that I can modify Rufus to detect it automatically. If it is an USB hard drive, then please let me know so that I can close the call.

Remember, the more __detailed__ information about what you are trying to do, and what you are using, the easier it is for someone else to try to help.

Danialde4, 2015-03-11:
Yeah. My problem is solved.

Ok.

Here's the log:

---
Rufus version: 2.0.639
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
0 devices found
0 devices found
0 devices found
0 devices found
Found USB 2.0 device 'Generic USB Disk USB Device' (8087:0024)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 9729, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 3
Disk ID: 0xF01A7C4A
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 45.2 GB (48566894592 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2:
  Type: NTFS (0x07)
  Size: 24.4 GB (26213351424 bytes)
  Start Sector: 94859264, Boot: No, Recognized: Yes
Partition 3:
  Type: FAT32 LBA (0x0c)
  Size: 4.9 GB (5242880000 bytes)
  Start Sector: 146057216, Boot: No, Recognized: Yes
Found USB 2.0 device 'Generic USB Disk USB Device' (8087:0024)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 9729, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 3
Disk ID: 0xF01A7C4A
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 45.2 GB (48566894592 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Partition 2:
  Type: NTFS (0x07)
  Size: 24.4 GB (26213351424 bytes)
  Start Sector: 94859264, Boot: No, Recognized: Yes
Partition 3:
  Type: FAT32 LBA (0x0c)
  Size: 4.9 GB (5242880000 bytes)
  Start Sector: 146057216, Boot: No, Recognized: Yes
---

I'm using external HDD. Before this I'm using pendrive and it's worked flawlessly.

And, thanks a lot for helping me sir.

May God bless you.

PS: I'm sorry if my English bad, cause I'm not native English.

pbatard, 2015-03-11:
OK, I'll close this issue then.

> PS: I'm sorry if my English bad, cause I'm not native English.

Don't worry. I'm not a native English speaker either :wink:

closed 2015-03-11 by pbatard