pbatard/rufus #461: Windows could not prepare the computer/error 0xc0000005 when installing Windows in UEFI/GPT/NTFS mode

pbatard, 2015-03-11:
This issue was [reported by Mr.X on forums.mydigitallife.info](http://forums.mydigitallife.info/threads/50859-Update-Rufus-USB-Tool/page10?p=1003590&viewfull=1#post1003590).

Basically, if you try to install Windows x64 in UEFI/GPT/NTFS, you will get either the message (removable disk):
```
Windows could not prepare the computer to boot into the next phase of installation. To install Windows, restart the installation.
```
or (fixed disk):
```
Windows installation error 0xc0000005
```
when Windows setup tries to start copying the files.

This is due to Rufus setting the type of the UEFI:TOGO partition to EFI System, and Windows setup creating another EFI System partition, but expecting only one EFI System partition for the whole system.

The reason Rufus chose to set the UEFI:TOGO partition to EFI System was for increased EFI compatibility, but clearly this wasn't a good choice. Will need to revert the type of the partition to Basic Data, and hope that all UEFI firmwares can list and boot from a partition that contains the expected EFI boot loaders, regardless of its type.

closed 2015-03-11 by pbatard

wmark, 2015-03-15:
I am getting this with UEFI/GPT/FAT32.

```
Expert command (? for help): i
Using 1
Partition GUID code: EBD0A0A2-B9E5-4433-87C0-68B6B72699C7 (Microsoft basic data)
Partition unique GUID: 1F0754F9-A2C5-4FC7-BC04-415DCAD9A885
First sector: 2048 (at 1024.0 KiB)
Last sector: 30998494 (at 14.8 GiB)
Partition size: 30996447 sectors (14.8 GiB)
Attribute flags: 0000000000000000
Partition name: 'Microsoft Basic Data'

Expert command (? for help): p
Disk 1:: 30998528 sectors, 14.8 GiB
Logical sector size: 512 bytes
Disk identifier (GUID): C300D58E-2855-4BFB-9FC9-26AE5579831B
Partition table holds up to 128 entries
First usable sector is 34, last usable sector is 30998494
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048        30998494   14.8 GiB    0700  Microsoft Basic Data
```

pbatard, 2015-03-15:
And?

Also, please don't __hijack__ an issue that pertains to a different problem(?) to report something else.
The original issue only applies to UEFI/GPT/NTFS. If you're not using UEFI/GPT/NTFS __and__ have an issue during installation, you should create a new issue.

cookieboyeli, 2015-03-24:
Hi pbatard, I believe this actually fixed the Windows To Go booting issue I was having in #450.

I didn't remember the specific error message until it was mentioned. 
"Windows could not prepare the computer to boot into the next phase of installation. To install Windows, restart the installation." was actually what I was getting after booting to the Windows To Go installation. (UEFI/GPT/NTFS) I was able to boot successfully without error after formatting with 2.1.649.

Elemshalit, 2016-03-13:
Hello,

I was wondering if someone can shed some light how to revert the partition to basic data on OSX. I am getting the same error message " Windows could not prepare the computer to boot into the next phase of installation. To install Windows, restart the installation. " Do I need a windows computer for this? 
Thanks! Thanks!

Elemshalit, 2016-03-14:
Ok. Fixed the issue by partitioning the drive to empty on yosemite and formatting the drive again with the windows installer. I am using bios UEFI. Anyway, thanks!

commit 0905e932 (2015-03-11): [core] set UEFI:TOGO GPT partition type to Basic Data

* It was set to EFI System, however the Windows installer fails if it
  detects 2 EFI system partitions on the same system (and it creates
  one during standard installation).
* This caused UEFI/GPT/NTFS installation to fail with Rufus 2.0
* Closes #461
* Also add the missing copyright notices for Grub4DOS and GRUB2