pbatard/rufus #463: Unable to format NTFS (USB flash problem)

dackko, 2015-03-14:
Hello,

I am new here, so please don't attack me if i do something wrong.

I'm having some problems with my USB Flash, i can't format it to NTFS with any kind of GUI.
However, i can format my USB Flash to NTFS if i use Command prompt:
convert X: /fs:ntfs
or
format X: /fs:ntfs

So the problem now is that even Rufus is can't format the USB Flash and the installation of the OS on the USB is impossible.
[image]

My question to you guys, can this problem be bypassed? (because i'm sure i'm not the only one with a faulty USB Flash)

Thanks.

pbatard, 2015-03-15:
It looks like your drive is defective, so please run a bad blocks check (which Rufus can do).

dackko, 2015-03-15:
I've done that, it passed bad blocks check, with 2 passes, without any errors.

pbatard, 2015-03-16:
OK, then please provide the __FULL__ log from Rufus after trying to format through NTFS (after unchecking `Create a bootable disk using...`.
Also what happens if you try to format FAT32 or UDF instead of NTFS?

pbatard, 2015-03-23:
No update from OP &rarr; Closing.

closed 2015-03-23 by pbatard

sdao85, 2016-10-16:
I have the same problem. There is no bad check. Rufus is on the latest version. Below is the log:

Rufus version: 2.11.995
Windows version: Windows 10 64-bit (Build 14393)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta3
System locale ID: 0x0409
Will use default UI locale 0x0409
Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 65270, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x05A4A539
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 500.0 GB (536870831616 bytes)
  Start Sector: 63, Boot: Yes
Partition 2:
  Type: Rufus Extra (0xea)
  Size: 31.5 KB (32256 bytes)
  Start Sector: 1048575906, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '10X64.PRO.OFF16.en-US.Oct2016'
  Size: 7742980096 bytes
  Uses: EFI
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: 10X64.PRO.OFF16.en-US.Oct2016.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 63 sectors
Deleting partitions...
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
Format error: Undetermined error while formatting.
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (0011:7788)
1 device found
No volume information for drive 0x81
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 65270, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x05A7B0D3
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 500.0 GB (536869863424 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt

pbatard, 2016-10-16:
Can you format the same drive to NTFS using Windows? This looks like a Windows issue.