pbatard/rufus #464: Can't select FAT and MS-DOS

caldwelljt2, 2015-03-15:
I am ultimately trying to get ESXi 5.5 to boot and run on a GA-MA78GM-US2H (rev. 1.0) using a 8GB flash disk thumb.

The person who gave me this PC, gave me a USB stick that (started) to boot ESX4 off USB, I have no details on how this was accomplished, but it hung just after saying SYSLINUX, so I know it was making it at least far enough to start reading the USB disk contents. Since it didn't work, I continued on with my plan to use ESXi 5.5 (I have an intel nic from the HCL as backup should it not recognize the on-board one)

I tried Rufus at first, but it simply wouldn't boot (hanging just after post indefinitely). Since this motherboard does not have the latest bios, and since I read that this motherboard appears to only want FAT(16?) I tried a few other things and ran into a snag that I can't get around.

1. I first tried to use FAT and FreeDOS, and it didn't boot (same kind of response)
2. I then tried to use FAT and the iso image, same response
3. Finally I tried to use FAT and MS-DOS, but ran into a snag that I can't figure out

It gives me an error ""MS-DOS cannot boot from a drive using a 64 kilobytes Cluster size. Please change the Cluster size or use FreeDOS." However, When selecting MS-DOS and FAT (not FAT 32), the only available option is 64kilobytes. Is it just not possible to use Rufus to do both MS-DOS and FAT (not FAT32)?

I was able to format the drive using FAT16 in a windows command line, but in windows 8.1 it appears to lack the old /s option to copy system files. So I can't seem to make this disk to test to see if it works. My hope was to either get a working ESXi boot disk or failing that to try it again after updating the bios. I need to boot a USB DOS flash disk to flash it, which according to that resource above requires a FAT16 partition.

I'm open to suggestions beyond getting those two options to work. Other manual methods, command options, or even other tools. I would probably install a CD-ROM and/or hard disk to test this if I could but I don't have them on hand where I am at.

pbatard, 2015-03-15:
Your drive is too big. MS-DOS is indeed not compatible with 64K clusters, and past a certain drive size, 64K is the only size you'll be able to use.

Please try to find a smaller USB Flash Drive.

caldwelljt2, 2015-03-16:
pete: Is there no way to force the use of only part of the drive? i.e., i'd
like to force 32k and either forgo the extra space or format it as another
volume. If I can't force it with rufus, is there a way to format it prior
to using rufus and then have rufus use whatever partion space is available?

I've looked all over my place to find a smaller USB, but I just no longer
have anything smaller than 4gb that doesn't have other problems (either
isn't functional, or is too small).

pbatard, 2015-03-16:
You should be able to [create a VHD](https://github.com/pbatard/rufus/wiki/FAQ#How_do_I_create_a_VHD_drive_to_use_with_Rufus) with a size that can accommodate 32k or smaller sectors. Then you should be able to use the DD mode of Rufus to write it to USB.

caldwelljt2, 2015-03-22:
Pete:

Excellent suggestion. I was able to work around the issue this time by
using a different solution, although I encontered yet another error I
couldn't explain.

I used a CF to PATA converter, and first stuck the 1GB CF card into a USB
reader. I then used rufus to apply an image, that was able to boot via the
pata (it would not, however, boot in the USB reader directly). After
booting, it was unable to install the image to the CF card itself (then
being seen as a 1gb PATA drive). It did not say it was too small, but
complained at partition time that it was trying to create something outside
the barrier of the drive (I assumed it was trying to create something
slightly larger than the 1gb limit of the cf card?)

By accident, during this process, I had a 4gb USB drive inserted into the
system when booting, and found it was detected as an installation
candidate. i was able to use the 1GB CF/PATA combination as a sort-of
install "CD" for the 4gb USB, and it was able to boot without issue from
then on.

I do not know why this motherboard had such a difficult time booting almost
any USB I created, but for whatever reason, ESXi was able to correctly set
itself up and boot normally.

Thanks again for your feedback. Even though I didn't use it here, I have no
doubt that the concept is something I will apply to other issues in the
future.

-JonC

pbatard, 2015-03-23:
OK, I'll close this issue then.

closed 2015-03-23 by pbatard