pbatard/rufus #465: err : Could not copy files to target drive.

Roflbobl, 2015-03-15:
Hey! I want to update my bios, but when i'm using rufus on my usb, it says " Could not copy files to target drive". 

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Volume has an unknown Partition Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system: Task 1/5 completed
Creating file system: Task 2/5 completed
Creating file system: Task 3/5 completed
Creating file system: Task 4/5 completed
Creating file system: Task 5/5 completed
Format completed.
Writing master boot record...
Drive has an unknown Master Boot Record
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume GUID \\?\Volume{3f1f0bf5-149d-43c0-b514-9741f7dedf0d}\
Opened drive \\?\Volume{3f1f0bf5-149d-43c0-b514-9741f7dedf0d} for write access
Writing partition boot record...
Using FreeDOS FAT16 partition boot record
Confirmed new volume has a FAT16 boot sector
Failed to flush F: [0x00000001] Incorrect function.
Successfully remounted Volume{3f1f0bf5-149d-43c0-b514-9741f7dedf0d}\ on F:\
Copying DOS files...
Unable to create file 'F:\COMMAND.COM': [0x00000057] The parameter is incorrect.
.
\\?\Volume{3f1f0bf5-149d-43c0-b514-9741f7dedf0d}\ was already mounted as F:\
Re-mounted volume as 'F:' after error

Found USB 2.0 device 'USB 2.0 Flash Disk USB Device' (0204:6026)
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 63, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00052F2A
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 496.7 MB (520879616 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Closed Windows format prompt

pbatard, 2015-03-15:
Please always try to provide the __FULL__ log, and not just the part about the formatting, as it provides critical information about the version of Windows and other stuff.

Can you please run a bad blocks check? Copy errors tend to indicate a defective drive...

pbatard, 2015-03-23:
No update from OP &rarr; Closing.

closed 2015-03-23 by pbatard