pbatard/rufus #466: Rufus does not like Windows 98 / Windows 98 SE bootable ISOs?

andreas--e, 2015-03-15:
Well, I've tried so many of them now, but Rufus simple does not accept them.
e. g. 
https://staff.washington.edu/curreri/windows-98-boot-cd-download-this-iso/

http://www.allbootdisks.com/downloads/ISO/AllBootDisks_ISO_Image_Downloads25/Win98SE_bootdisk.iso

Rufus will _always_ complain that the ISO is neither based of winpe/bootmgr nor ISOLINUX.
Great.
WinPE is mainly used for XP/7/Vista/8 but _not_ for Win98/Win98SE.
Still, Win98/Win98SE still ARE the perfect choice for flashing BIOSes etc. Consider that most of the flashing tools themselves are for DOS.

Would there be any way to convince Rufus to write Win98/Win98SE to a pen drive?
Rufus is a great tool pointing in the very right direction - but when it comes to 98/98SE, you will try in vain. I've also scoured the net about this and dozens of users have run into this issue, ending up with manually copying 98/98SE or using other tools.
I can't believe that Rufus and its author like to block this important functionality just by relying on their stupid WinPE thing.

pbatard, 2015-03-15:
> I can't believe that Rufus and its author like to block this important functionality just by stubbornly relying on their stupid WinPE thing.

And I can't believe that some people don't understand that developers who aren't paid for it, and provide software to them absolutely for free, don't want to waste their limited time supporting an Operating System that is completely obsolete by now, instead of using that same time to, say, [improve accessibility for visually impaired persons](https://github.com/pbatard/rufus/issues/243), something that __will__ benefit a lot more people than the dozen or so people who, in 2015, think that using Windows 98 is a good choice. There is such thing as prioritization and not trying to add every single possible feature under the sun.

Besides, you __do__ realize that Rufus does allow you to create either a FreeDOS or MS-DOS boot disk to run your DOS flash tools. Why on earth would you want to use Win98 for DOS based tools? Win98 is NOT the perfect choice for flashing BIOSes. DOS is. And Rufus will happily created  a DOS bootable disk for you.

Closed as not_gonna_happen.

closed 2015-03-15 by pbatard

Sopor-, 2015-03-15:
@andreas--e  In 2015 you don't need DOS to flash... Try read the manual and you will find that you can put your USB-stick and flash it from BIOS/UEFI without using any OS at all.

andreas--e, 2015-03-17:
NOTE: When I mean "Windows 98" I mean DOS7 that comes with it. I never planned to *run* the GUI of Win98 with that, nor would I ever want to do so in future. I only need console mode.

@Sopor- Wish it only was so easy!!
It was an old board (sporting USB 1.1) that I wanted to flash and booting from USB was NOT possible. Or at least it did not accept the pen drive for booting, whilst---read on---this worked perfectly on my main PC.

Thanks to Rufus and its developer, I could make it happen, but NOT with a bootable disk.

- (1) Used Rufus to make a bootable FreeDOS "disk" on my pen drive, which I could NOT boot on the PC with the old board. OK.

-> Next task to solve: Get the data (flash tool + BIN file) off the USB just by accessing it from DOS. <--

- (2.1) Booted up with "Hiren Boot Disk" CD-ROM into DOS mode.
. (2.2) Selected Dos->USB Support->WITH EMM386.
- (2.3) Selected 1st option (Panasonic) *AND* UHCI for old PCs, as we're not on USB 2.0, but 1.1.

-> Success! Pen drive recognized and assigned to S: 

- (2.4) cd'ed to S: and flashed BIOS. HOORAY!

So Rufus did help me after all, just had to find a way how to get access to the data on the FreeDOS-formatted USB drive. Thank goodness FreeDOS is readable under MS-DOS as well.

TheArcaneBrony, 2017-10-20:
How about installing win98 for legacy apps, or at least some nostalgia/fun?