pbatard/rufus #467: Rufus suggest to upgrade grub and makes unbootable stick of kav rescue disk

gggeek, 2015-03-15:
When trying to create a usb stick for kaspersky rescue disk, rufus 2.0 prompted me to download grub 2.0 because only 2.0-beta was found.
The resulsting stick would not boot and drop me to a grub emergency prompt

pbatard, 2015-03-15:
I replicated the issue. Unfortunately, the Kaspersky people seem to be fudging their versioning of GRUB, as they explicitly set their version of `0`. 

This is __VERY BAD__ practice, and of course it makes it next to impossible for Rufus to figure out what is the __actual__ version of Grub that Kaspersky are using, so that their ISO can be converted to bootable USB.

At this stage, I may very well add code that states that the Kaspersky ISO (or any other ISO that sets its GRUB version to something as inaccurate as `0`) is unsupported...

gggeek, 2015-03-15:
I see. Otoh Unetbootin does not seem to have any problems with it.
Is the approach taken by the two apps completely different?

pbatard, 2015-03-16:
Yes it is. If I were to take a guess, I'd say Unetbootin probably tries to replace all of the GRUB files, whereas Rufus only tries to replace the GRUB bootloader.

pbatard, 2015-03-17:
Well, I spent the last 3 hours trying to guess the GRUB version Kaspersky used, but none of the one I tried (starting with 1.97) worked.
So Kaspersky or any ISO that can't report a proper version is out.

commit ee5af9f8 (2015-03-17): [grub] disable support for ISOs that don't report a proper GRUB version

* Also update localization for supported boot methods
* Closes #467
closed 2015-03-17 by pbatard