pbatard/rufus #469: rufus 2.0 fails when burning iso image to 32GB USB drive

jzoudavy, 2015-03-17:
Hi 

I got a Kingston Data Traveler 3.0 USB Device and I am using RUFUS 2.0 to burn OpenSUSE 12 ISO and 13 ISO. The process fails with error ISO Image extraction failure. It worked with the ESX 5.5.2 ISO. 

[image]

The log is attached, I have changed the extension from .log to .jpg to get around the problem that I can't seem to attach anything that's not a picture. 

Thanks

pbatard, 2015-03-18:
Please copy/paste the log. I can't see it at all.

pbatard, 2015-03-18:
Nevermind, I got it (and fixed your entry). But if you need to work around extensions, you are doing something wrong. It's a text file. You can copy/paste text, and it is the preferred method to pass on information so that everybody can see it.

pbatard, 2015-03-18:
According to the log, your ISO could be corrupted. Can you please confirm that the sha1sum of your `openSUSE-13.1-DVD-x86_64.iso` is [`0b8c40a36ff514797cb2ef69331f1c3c8a9ec6ca`](http://download.opensuse.org/distribution/13.1/iso/openSUSE-13.1-DVD-x86_64.iso.sha1)?
Or you can use the [md5sum](http://download.opensuse.org/distribution/13.1/iso/openSUSE-13.1-DVD-x86_64.iso.md5) if you prefer.

I'll try to download and test the image, but this could take a while, and you should always try to confirm that your ISO is not corrupted when you encounter a copying issue in Rufus.

pbatard, 2015-03-18:
Tested openSUSE-13.1-DVD-x86_64.iso. Found no issue during the USB creation:
```
Rufus version: 2.0.639
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Using autorun.inf label for drive F: 'openSUSE 13_1'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005CBE7E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 111.8 GB (120033075200 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
[... 1102 lines cut ...]
Extracting: F:\suse\x86_64\zoo-2.10-1019.1.2.x86_64.rpm (61.4 KB)
Extracting: F:\suse\x86_64\zsh-5.0.2-8.1.4.x86_64.rpm (2.5 MB)
Extracting: F:\suse\x86_64\zypper-1.9.3-1.1.x86_64.rpm (940.1 KB)
Extracting: F:\suse\x86_64\zypp-plugin-python-0.3-10.1.2.x86_64.rpm (5.2 KB)
Extracting: F:\SuSEgo.ico (2.2 KB)
Created: F:\syslinux.cfg
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Using autorun.inf label for drive F: 'openSUSE 13_1'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x005FAA5B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 111.8 GB (120033075200 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```
The issue is either that your ISO is corrupted or that your drive is defective. Please check the sha1sum from your ISO and also try to run a bad blocks check (which Rufus can do) against your drive.

For good measure, I also tested `openSUSE-12.3-DVD-i586.iso` and found no issue.

jzoudavy, 2015-03-18:
hey Pete,

thanks for your deligience in this matter. I have confirmed that our ISO is
indeed corrupted.

You mentioned that the logs showed the corrupt ISO, how so? Was it these
lines:

Extracting: F:\directory.yast (309 bytes)
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order

Perhaps this can be new feature in the future? auto check of iso for
integrity?

On Tue, Mar 17, 2015 at 11:11 PM, Pete Batard <notifications@github.com>
wrote:

> Tested openSUSE-13.1-DVD-x86_64.iso. Found no issue during the USB
> creation:
>
> Rufus version: 2.0.639
> Windows version: Windows 8.1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub [...]

pbatard, 2015-03-18:
> You mentioned that the logs showed the corrupt ISO, how so? Was it these lines:
> Extracting: F:\directory.yast (309 bytes)
> libcdio: from_733: broken byte order 
> libcdio: from_733: broken byte order

Nah. Whilst annoying (and not something I can easily fix since they come from libcdio), the `broken byte order` messages are benign.

What told me that your ISO might be corrupt was the read error at the end of your log:
```
Extracting: G:\docu\RELEASE-NOTES.en.html (21.8 KB)
  Error reading ISO9660 file /docu/RELEASE-NOTES.en.html at LSN 2232287
```
If Rufus can't read a file from the ISO, one of the first guesses is that the ISO can be corrupt.

> Perhaps this can be new feature in the future? auto check of iso for integrity?

That would be __very__ difficult to do.
All Rufus gets when a user selects an ISO is the ISO name. It doesn't tell where that ISO was downloaded, or whether a checksum is available. And even if we could try to take a guess as to where the ISO originated, different sites will present the checksum differently (some in separate files, some in the page itself). Also if the user has renamed their ISOs all bets are off!

The only possible place to check if an ISO is corrupt is at download time. But Rufus is not being used for the download, so what you are asking is next to impossible to do I'm afraid...

I will close this issue now.

closed 2015-03-18 by pbatard