pbatard/rufus #470: Support for 32 bit UEFI in Rufus 2?

riklaunim, 2015-03-17:
In Rufus 1.4 I could burn Linux ISO, copy bootia32.efi to EFI\Boot and it would work with weird Bay Trail laptops/tablets with 32-bit UEFI. With Rufus 2 the file structure is totally different and now I can't make anything that will boot on a Win 8.1 Bay Trail tablet (UEFI sees the USB device but doesn't boot at start even when selected as the device to boot from). Is there a way to make it working for such devices?

pbatard, 2015-03-18:
Rufus shouldn't have changed the copy process for 32 bit ISOs. Can you please provide a link to the Linux ISO you are using (it's hard to test if you don't tell me what you are actually using).

Also a __full__ log from Rufus 1.4.12 and 2.0, after you create the USB from your ISO, would be __invaluable__, as comparing its data should tell precisely if there are any differences in the copy process for `bootia32` (or any other issue for that matter).

riklaunim, 2015-03-18:
Will do it later today. I was using Ubuntu 14.10 and it derivatives before.

pbatard, 2015-03-18:
I've just downloaded `ubuntu-14.10-desktop-i386.iso` and I can see that, unlike the x64 version, it does __NOT__ contain anything allowing it to boot in UEFI mode.
To be EFI boot compatible, it should have an `EFI/boot/bootia32.efi` file, but it doesn't have any.

So I gather you're trying to copy a __non Ubuntu approved__ `bootia32.efi` file (since it isn't part of that image) and finding that your custom workaround to make your USB bootable does not work...

Sorry, but I can only provide support for EFI boot if the ISO is EFI boot compatible in the first place.
If it isn't, and you're trying to use your own custom process to make it EFI bootable, then I'm afraid you're on your own as this is not a Rufus issue.

Unless you can point to a 32 Linux ISO, that is EFI bootable, and that doesn't work when Rufus converts it to USB, I will now close this issue.

closed 2015-03-18 by pbatard

riklaunim, 2015-03-18:
It was for booting 64 bit Ubuntu on 64 bit CPU on a device with 32 bit UEFI.

pbatard, 2015-03-18:
`ubuntu-14.10-desktop-amd64.iso` only provides x64 EFI boot files. It does not provide x32 EFI boot files.
So that's not gonna work with 32 bit UEFI either, as it requires a very specific `bootia32.efi` file.
Also, unlike what is the case with Windows x64, you cannot use 32 bit EFI code on 64 bit EFI systems.

So really, if you boot 32 bit UEFI, you must have a `bootia32.efi` on your ISO somewhere. If your ISO doesn't provide that, it will be unsupported in Rufus.

riklaunim, 2015-03-18:
I'll re-check it tonight. With Rufus 1.4 I had EFI/boot folder in which that bootia32.efi could go and make it bootable. Maybe I used the wrong ISO now with Rufus 2. I'll recheck.

A lot of BayTrails has that annoying 32 bit UEFI that in general isn't very standard and is not supported by Linux distributions. Current Debian/Ubuntu 15.04 was working on supporting this unobvious UEFI, but I don't know if and what are the results of that.

pbatard, 2015-03-18:
> A lot of BayTrails has that annoying 32 bit UEFI that in general isn't very standard and is not supported by Linux distributions.

Yes, I know... :disappointed: 
But I tried to make sure that Rufus would be compatible with 32 bit EFI bootable ISOs (and it seems to work fine with Windows), so it should work with such a Linux ISO if you can find one.
If it doesn't, and that ISO does contain a `EFI/boot/bootia32.efi` file, I definitely want to know!

riklaunim, 2015-03-18:
I've re-checked. The x86_64 ISOs do work as they should so I was using x86 iso which doesn't support UEFI. So now Rufus made flash drive boots on a tablet (although it can only display on HDMI ;))