pbatard/rufus #471: ESXI 6 ISO

atkin05, 2015-03-18:
When trying to open the ESXI 6 ISO I get an error: 
This version of Rufus only supports bootable ISOs based on bootmgr/WinPE, ISOlinux or EFI. This ISO doesnt appear to be either.
Would it be possible to add support for what ever the ESXI 6 ISO is using?

pbatard, 2015-03-18:
I'd like nothing more than investigate issues with ESXi 6... if I could get my hands on an ISO. I subscribed to the ESXi 6 BETA on the VMWare site, but I never got anything back with regards to testing...

atkin05, 2015-03-18:
I have the trial ISO. Do you have somewhere I could upload it? It's available free from VMware and lasts 60 days.

pbatard, 2015-03-18:
Finally got it from the VMWare site.
What version of ESXi 6.0 are you using?

I got the trial `VMware-VMvisor-Installer-6.0.0-2494585.x86_64.iso` off the VMWare website, and I have no issue opening it. That image seems to have both EFI and Syslinux boot support, and I was able to create the USB without trouble...

pbatard, 2015-03-23:
No update from OP &rarr; Closing.

closed 2015-03-23 by pbatard

atkin05, 2015-03-25:
> No update from OP → Closing.

VMware-VIMSetup-all-6.0.0-2562643

pbatard, 2015-03-25:
Well, that ISO doesn't seem to be bootable at all. have you checked that you could __actually__ boot it using an optical drive if you write it to a DVD? Especially, it doesn't even seem to use El Torito, let alone any other boot method. I tried to boot a VM using that ISO, and it just doesn't...

Rufus will not magically make an ISO bootable if it can't be booted from the CD or DVD in the first place...

atkin05, 2015-03-25:
Thats really odd, i revisited the same link in "my vmware" and it's change
the downloader, it's working now.