pbatard/rufus #472: Error: Media cant be accessed

seeker-seek, 2015-03-19:
I am trying to create a bootable USB using Rufus.

My error log is like this

Rufus version: 2.0.639
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1953, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.0 GB (16063135744 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: Win81-x64-Feb2015
  Size: 6537598976 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win81-x64-en-US-Mar2015.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'I:' as volume mountpoint
I/O boundary checks disabled
Could not get exclusive access to device \\?\Volume{f46fef74-7104-11e3-824b-e7c22a5db7bb}: [0x00000005] Access is denied.
Could not lock volume
Re-mounted volume as 'I:' after error

Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1953, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.0 GB (16063135744 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (058F:6387)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1953, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 15.0 GB (16063135744 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-03-19:
Are you using any of the software mentioned [here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) by any chance?

pbatard, 2015-04-02:
No update from OP. Closing.

closed 2015-04-02 by pbatard