pbatard/rufus #475: Add verification sums to advanced options.

Sinnlosvoll, 2015-03-23:
It would be really nice to view sha/md5-sums when an iso is selected.

I don't know about the performance impact that has, maybe make it a checkbox to calculate these sums.

pbatard, 2015-03-23:
You know what, I was actually thinking of adding an enhancement request for this. However, I don't want to do it all the time (or every time the user selects the advanced mode) as hashing a 8GB ISO can be very time consuming. It usually takes more than a minute to do so even with a fast computer and from an SSD (and can of course be a lot longer if you have a slow CPU and HDD).
Even doing it in the background, and displaying the sums once completed is not something I'd want to do, as having the app use a lot of CPU might not be something users want.

Moreover, usually you only want the md5sum or the sha1sum, but rarely both. Computing both is nice of course, but takes some extra time.

So the way I see it right now is I'll probably implement this as a cheat mode with something like `Alt-M` to compute the md5sum and `Alt-H` (`Alt-S` is already used for something else) to compute the sha1sum. Then Rufus would display the computed sum in the info field (under the progress bar), as well as in the log.

How does that sound?

Sinnlosvoll, 2015-03-23:
That sounds awesome!
Maybe a database-file for already calculated hashes would also be beneficial.

closed 2015-06-29 by pbatard

Walkman100, 2015-06-30:
Awesome! I've recently been doing a bit of Hash calculation myself, I have MD5, SHA1 and SHA256 sum calculation to my [Properties](https://github.com/Walkman100/PropertiesDotNet) program.
> hashing a 8GB ISO can be very time consuming

Ahhhh - that makes a lot of sense. I was wondering why it took a few milliseconds to hash my test files.
> [Commit Msg] Uses the Alt-M cheat mode for now

Does it calculate both on <kbd>Alt</kbd> + <kbd>M</kbd>?

pbatard, 2015-06-30:
> Does it calculate *both* on <kbd>Alt</kbd> + <kbd>M</kbd>?

It does. :smile: My expectation is that since this is a time consuming operation anyway, and that regular joes might not know whether they need the SHA-1 or the MD5, we might as well do both at the same time (as it'll avoid re-reading the same image twice). If I have idle time on my hands, and it looks like many people are using the feature, I might even look into ways to parallelize operations if you have more than one core, to speed things up further. Oh, and I'm also trying to devise a way to have the hash computation initiated from some kind of button on the UI (rather than a keyboard cheat mode) and the hashes popping out in a separate dialog to make it easier to copy/paste them than from the log, where they are generated now.

PS: Nice one with the `<kbd>` and `</kbd>`. I didn't know that markup, but I'm sure I'll be using it all the time now... :+1:

commit c95910e2 (2015-06-29): [misc] add MD5 and SHA-1 computation for current image

* Uses the Alt-M cheat mode for now
* Closes #475