pbatard/rufus #476: Cannot mount GUID volume

cookieboyeli, 2015-03-24:
Hey I got an error when trying to format back to a bootable windows install from a Windows To Go install. I tried a few times, realized it was in a USB 2.0 port, plugged it into USB 3.0, got the error a few more times and then it worked when I used FAT32 instead of NTFS.

Here's the full log:
```
Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {C7B07408-B0DB-4CFF-9B2C-571BCADEE934}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {97B22635-CA32-4B1F-AECB-3C5071AFB82E}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {8C7C1C2C-D5A5-4958-908E-832F72F23F3D}
  Size: 29.0 GB (31136555520 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {9F0BDD4E-8063-4647-A051-B91E036C831E}
  Size: 100 MB (104864256 bytes)
  Start Sector: 61077777, Attributes: 0x0000000000000000
Scanning image...
Disc image is an UDF image
ISO label: Win81-x64-Feb2015
  Size: 6537598976 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win81-x64-en-US-Mar2015.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
[... 1162 lines cut ...]
Extracting: G:\support\logging\setupetw.dll (8 KB)
Extracting: G:\support\logging\setupugcetw.dll (4 KB)
Extracting: G:\support\logging\sysprepetw.dll (6.5 KB)
Extracting: G:\support\logging\windeployetw.dll (3.5 KB)
Extracting: G:\support\logging\winsetupetw.dll (6.5 KB)
Finalizing, please wait...
G:autorun.inf already exists - keeping it

Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 2.0 device 'SanDisk Extreme USB Device' (0781:5580)
NOTE: This device is an USB 3.0 device operating at lower speed...
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {A6DEF0C0-A631-4047-BD28-DC80D84C0829}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {01984AD8-596E-4645-91FA-31583D345831}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2015-03-25:
I tried to replicate your issue (same initial conditions with a To Go drive, and using an USB 3.0 drive on an USB 2.0 port), but no luck so far...

cookieboyeli, 2015-03-25:
Ok I will try to replicate it again and mark down the exact steps I took if I can replicate it.

pbatard, 2015-03-25:
The weird thing is you seem to have had 2 partitions mounted on that original drive, which shouldn't happen (even for a fixed drive) unless you forced it.
For instance, at least in Windows 8, you cannot mount the EFI System partition using the regular GUI tools, and you have to assign a letter manually through commandline if you want Windows to see it. And the MSR should be even less accessible.

So it really looks like you have a very unconventional setup when your drive was mounted, as Rufus detected that 2 drive letters were in use (`E:`, `G:), when you really should only have had the main NTFS partition mounted. But I also tried to mount the EFI System Partition on my system to have 2 drive letters on the drive before formatting, and still got the drive installed as expected,

cookieboyeli, 2015-03-25:
I'm not sure which dive you're referring to when you say "The weird thing is you seem to have had 2 partitions mounted on that original drive, which shouldn't happen (even for a fixed drive) unless you forced it."

But I can tell you that I did screw something up. I had a windows install on a 256GB SSD (C:) and another windows install on a 120GB SSD (D). I also have a 500GB HDD that assigned a letter much further down the alphabet for some reason. I reformatted and reinstalled windows on the 256GB while the 120GB was plugged in and apparently some things that would normally go on the 256 went on the 120 (I think this is the 2 partitions you are talking about). So I have to reinstall with the second windows install disconnected to avoid this. Also, if I format the 120GB or unplug it I cannot boot to my main 256GB.
Sorry if that's completely irrelevant, I just don't have enough experience in this area yet.

I haven't had a chance replicate yet but I will within an hour.

cookieboyeli, 2015-03-25:
Ok I can still replicate it. But I think it's something specific to my setup as you said.

All I have to do to get the error is try formatting GPT:UEFI:NTFS
MBR:UEFI:NTFS works fine.

I am getting an error on exFat SOMETIMES stating my volume label is invalid, but sometimes it works. ???

Here's the log:
```
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {37670637-4105-414B-8E05-09932A8D29F5}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {23B45F6B-E32D-465C-B897-B0A071BFE79F}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {37670637-4105-414B-8E05-09932A8D29F5}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {23B45F6B-E32D-465C-B897-B0A071BFE79F}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume does not have an x86 Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (ReFS)...
Using default cluster size
Quick format was selected
Format completed.
Found volume GUID \\?\Volume{5863263a-e7b2-49a1-b0a1-5852b0085e5f}\
Volume is already mounted, but as E: instead of G: - Unmounting...
Created: G:\autorun.inf
Created: G:\autorun.ico
Successfully remounted Volume{5863263a-e7b2-49a1-b0a1-5852b0085e5f}\ on G:\

Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
[... 74 lines cut ...]
Disk GUID: {A788F99C-8265-413C-9109-888126ABB41B}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {84D2EFAA-2D1A-4102-91A8-C7D206079473}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {A788F99C-8265-413C-9109-888126ABB41B}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {84D2EFAA-2D1A-4102-91A8-C7D206079473}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

cookieboyeli, 2015-03-26:
So I just did some further testing with 2.0 and 2.1 and now I cannot replicate the issue anymore. GPT:UEFI:NTFS is working fine. I haven't even restarted my computer since then or changed any other storage media.

I don't know what to make of this. Here is the full log from the successful 2.1 format.
```
Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {695BAFDC-0D41-4451-A75C-5316E22B57F5}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {023270D7-CAC3-46A0-AE77-F5934652DD56}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{27f776e3-b424-4741-b5fc-511e343664c9}\
Volume is already mounted, but as E: instead of G: - Unmounting...
Created: G:\autorun.inf
Created: G:\autorun.ico
Successfully remounted Volume{27f776e3-b424-4741-b5fc-511e343664c9}\ on G:\

Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {0E3B4BCD-0F3E-428C-812A-7E2CF4C4E492}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {27F776E3-B424-4741-B5FC-511E343664C9}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive G: 'Win81-x64-Feb2015'
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {0E3B4BCD-0F3E-428C-812A-7E2CF4C4E492}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {27F776E3-B424-4741-B5FC-511E343664C9}
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

cookieboyeli, 2015-03-30:
Got it again!
```
  Size: 29.2 GB (31375641600 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
Disc image is an UDF image
ISO label: Win81-x64-Feb2015
  Size: 6537598976 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Win81-x64-en-US-Mar2015.iso
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Closed Windows format prompt
Creating file system...
Format completed.
Found volume GUID \\?\Volume{c8cc26ac-e3bd-4374-9b9a-48eb1320752c}\
Volume is already mounted, but as E: instead of G: - Unmounting...
G:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{c8cc26ac-e3bd-4374-9b9a-48eb1320752c}\, got \\?\Volume{7823b4f4-5754-4b5e-88c5-0f0b9eb91197}\
Could not remount \\?\Volume{c8cc26ac-e3bd-4374-9b9a-48eb1320752c}\ on G:\: [0x000000EA] More data is available.
G:\ already mounted, but volume GUID doesn't match:
  expected \\?\Volume{c8cc26ac-e3bd-4374-9b9a-48eb1320752c}\, got \\?\Volume{7823b4f4-5754-4b5e-88c5-0f0b9eb91197}\

Found USB 2.0 device 'Multiple Card  Reader USB Device' (058F:6366)
Ignoring autorun.inf label for drive F: No media
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
2 devices found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3814, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3568832C-1693-4FEC-92E7-FE84D138DADB}
Max parts: 128, Start Offset: 17408, Usable = 31376672768 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {C8CC26AC-E3BD-4374-9B9A-48EB1320752C}
  Size: 29.2 GB (31374572544 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {7823B4F4-5754-4B5E-88C5-0F0B9EB91197}
  Size: 256 KB (262144 bytes)
  Start Sector: 61280510, Attributes: 0x0000000000000000
```

pbatard, 2015-08-28:
I think I have an idea what's going. Will add a possible fix for this with the 2.3 release.

commit 3839bf42 (2015-08-28): v2.3 (build 709)

* Closes #476
* Also removes the stacksize warning during VS code analysis
closed 2015-08-28 by pbatard