pbatard/rufus #481: Can't select iso from network drive(s)

victorheld, 2015-03-29:
When i try to select an ISO from my NAS it says my NAS is unavailable but in other programs it works just fine.

    Rufus version: 2.1.649
    Windows version: Windows 8.1 64-bit

Error message:
[image]

pbatard, 2015-03-29:
This is as expected, because you must remember that Rufus needs to run as Administrator, and therefore will not see the shared drives that are mounted for a different user. Your regular user is not the same as the Administrator, therefore the network shares that these two users will see are different.

I have now added a [FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cannot_see_my_network_shares) about this (and now, it wouldn't be easy to work around this, and considering that the percentage of people accessing ISOs from network shares is expected to be low, it wouldn't make sense for me to invest time on this).

Closing issue.

closed 2015-03-29 by pbatard

Sopor-, 2015-03-29:
So you can't check if UAC is disabled? Or it has noting to do with UAC?

pbatard, 2015-03-29:
It has nothing to do with UAC:
1. To be able to format/partition a drive, and application __must__ run as elevated. There's no way Rufus can do anything if it runs as the regular user. And this is why you get an UAC prompt when launching Rufus.
2. This means that the actual user that will be running the application is `Administrator`.
3. If Rufus was to try to toggle back and forth between the regular user (for ISO selection) and Admin user (for everything else), which probably would incur more UAC prompts, I'd have to figure out the network drive mappings, which might be difficult, as well as __provide the credentials (username + password)__ to be able to duplicate this mapping as Admin. Except the application has no way to figure out these credentials (else, the whole Windows security scheme would be useless), so I'd have to prompt the user every time they want to access that share, since I'd be a security risk to keep a permanent mapping of these shares for Admin, and the last thing I want is for Rufus to make someone's computer less secure.

The only non prohibitively intrusive possible scheme I see to allow ISO selection from the regular user's mapped drives is if I embedded a separate app in Rufus, which would get launched under the regular user's credentials, when they click the image selection button, and that would somehow pipe the data as needed to the main application that is running as Admin. This would add A LOT of complexity, as well as too many potentials points of failure.

Again, for something that only a very small percentage of Rufus users are going to use, and which can __super easily__ be worked around by either copying the relevant files locally, it just doesn't make sense for me to put that much effort into it when I already have a lot more features to add with higher priority (because they can't easily be worked around), and when my development time is a lot more limited than people imagine...

adventiar, 2015-09-03:
I had the same issue. I worked around it by adding a network location. Unlike a network drive a network location is simply a shortcut to a share. Since you need to create a share for your network drive to point to you can just add a network location to the same share. Now you can access the share as an actual network drive or as network location. Once you open Rufus just go to "My Computer" ("My PC" if using windows 8 or later) and click the network location. If you don't know how to create a network location use the following link and look at the section below the "Note:" near the bottom of the page. it talks about adding webpages but just make the address \\server\share.