pbatard/rufus #482: Package portable version differently

RonS2015, 2015-04-04:
Hello Pete,

Love Rufus, what a wonderful utility! Thank you so much for making it available for free to everyone.

I think portable apps are great and carry around a whole bunch on USB drives. Thus far Rufus seemed to be almost but not quite portable, so I was overjoyed to see a separate download for such a version on the site. Unfortunately the app still wrote to the registry, to no surprise since on checking it was exactly the same executable as the non-portable version.

The changelog for 2.0 gave me a hint that portable app support required a 'Rufus.ini' file, and a quick Bing search gave me a link to [the file on GitHub](https://github.com/pbatard/rufus/blob/818fa4da88a600f4f25f95e85ecce9a53c00fd0e/res/rufus.ini).

I would suggest that this same INI file (explanatory comments included) be packaged with the executable in a ZIP archive and be offered for download as the portable version. As it stands having two separate links to the same EXE but calling one portable makes no sense, and from the main program page it is not clear what steps might need to be taken to get portable mode to work (unless I missed something obvious?).

Hope to see this implemented soon.

Thanks!

pbatard, 2015-04-04:
Please see [here](http://www.portablefreeware.com/forums/viewtopic.php?p=73455&sid=732e910dd43630202039ea559e487000#p73455).

As long as the application name contains a `p`, Rufus will create a `rufus.ini` file automatically, so there is no need to provide it with the download.

Basically, here's how the executable works:
1. Is there a `rufus.ini` file in the same directory as the one the app was launched from. If there is, start in portable one.
2. If there isn't a `rufus.ini`, check if the name of the executable contains a `p`. If it does, create a `rufus.ini` and start in portable mode.
3. If none of the above applies, start in portable mode.

Note however that, as explained in the sample `rufus.ini` provided [here](https://github.com/pbatard/rufus/blob/master/res/rufus.ini), if you don't want the app to write to the registry at all, you should add a `DisableLGP = 1` line in your `rufus.ini` (but then prepared to be pestered by Windows every now and then about unformatted drives).

I hope you do realize that _portable_ doesn't equate _not writing to the registry_.
Portability simply means that you can take the app to another computer and keep the same settings, and that's exactly how the portable mode of Rufus operates, through the `ini` file.

As I believe I answered your question, I will now close this issue.

RonS2015, 2015-04-05:
> As long as the application name contains a p, Rufus will create a rufus.ini file automatically, so there is no need to provide it with the download.

A-ha, looks like I shot myself in the foot there by renaming the **p**ortable version before running it. Sorry for the confusion and thanks.

JonSN, 2015-08-02:
The 'p' in the name seems to be case sensitive? Can you please make the check case insensitive so that I can name the executable as RufusPortable.exe (instead of the ugly Rufusportable.exe :)?

pbatard, 2015-08-02:
I think I can do that. But next time, __please__ create a new request instead of piggybacking on an old one.

commit e1f76a5d (2015-08-04): [core] make check for portable version case insensitive

* Closes #482
closed 2015-08-04 by pbatard