pbatard/rufus #483: Language button requires a tooltip

RonS2015, 2015-04-04:
That globe icon doesn't really make it clear what it's for. A tooltip would be nice.

pbatard, 2015-04-04:
I think that's a fair request.

I'll see what I can do, but this will require rallying 30 translators to translate the tooltip, so it will probably be a couple of months at best to get this enhancement added.

Halamix2, 2015-04-07:
Or change the globe to flag?

pbatard, 2015-04-07:
Rufus is very size conscious, so wherever it can, it uses icons from the Windows system rather than one embedded (Why? Because embedded icons have would have to scale nicely for high DPI displays, and a nicely scaling icon usually adds 100KB!).

IMO, the current globe + arrow icon is actually better at conveying that the button allows language selection than any of the other icons that are native to a Windows system. But I'm open to suggestions for a better icon if you can find one (NB: you can use the excellent [IconsExtractor from NirSoft](http://www.nirsoft.net/utils/iconsext.html) to poke system DLLs, such as `shell32.dll`, for icons)

Also, if I were to pick a flag, which flag should I pick? As pointed out, I cannot embed all 30+ flags for the languages Rufus supports, as it would probably increase the size of the app by half. And using only one flag would be confusing IMO...

pbatard, 2015-07-07:
Done in 91e699d38a5cb2ee66a9ef3d8c69d0d2cff40bb8.
Translator updates will be added when I finalize the list of new translations needed (#504), which will happen __after__ 2.3 is out.

closed 2015-07-07 by pbatard