pbatard/rufus #484: Disabling update check

RonS2015, 2015-04-04:
On launching the app for the first time it shows this:

[image]

If I click on More Information, it shows this:

[image]

Now on this More Info dialog if I disable updates, then IMO closing it should show the app UI directly. Currently one can disable updates on the More Info dialog, then when the first dialog is shown again, one can re-enable updates (perhaps by mistakenly pressing yes). So why not skip showing the first dialog again if updates have been explicitly disabled from the second (More Info)?

pbatard, 2015-04-04:
This might be tricky, and quite frankly I don't see it as that big a deal (of course you should expect the last action you perform to take precedence, and more info is mostly for the update policy).

I'll keep this open, but this will be __very low__ priority, and I may close this request at any time if I feel like it, as I really don't see this as that big a deal...

pbatard, 2015-04-04:
Also, users might actually be confused if they click more info and never come back to the initial dialog, as the expectation is that "more info" should only be a temporary diversion to the process of selecting whether one does or does not want update check from the first popup.

I'm pretty sure that, if I go with your request, I will get counter-requests from users complaining that they never come back to the first popup, and therefore are not sure what the setting is.

All things considered, I think I'm gonna leave it as is. Better leave users with _2_ chances of selecting whether they want to allow update checks than 1.

closed 2015-04-04 by pbatard