pbatard/rufus #485: Linux / Mac OS X fork

ivkol, 2015-04-06:
Hello Dear,

Appears github have changed their policy and no personal messages exist? WTF

What would you recommend as start point if I were to rewrite the application for Linux and MacOSX. Is it a huge hassle to handle block devices? should be much easier than on windows i guess.

Does such tool exist for Linux/Mac OS X already ?
I have tried lots of useless tools like unetbootin and such... noone of those works like your tool (always, from first time with every iso i feed it).

I have a problem creating Windows bootable USB flash drive and I am trapped in a useless loop of lots of work until I prepare a USB flash on my MacBook Pro.
 I dont have bare metal Windows around, neither my Virtual Machines on my Mac can export USB to themselfs properly so I can use Rufus(doesnt find devices).
Running Rufus on OSX with wine finds no devices as well.
I need to find a retail cd of windows 7, bootcamp it(not every iso works), install on baremetal, install rufus, download Windows 7 SP1 or any newer, and prepare usb with Rufus from baremetal windows machine.
So I always have to ask a friend to prepare a Usb drive for me or I need to have a bare metal windows machine around to do the hassle by myself.

Ofcourse I do 'dd if=/dev/disk2 of=~/img/win7.sp1.usb8g.img' images that can I restore with dd on 'any' other usb. 
But newer releases of upgraded windows exist also and I will like to prepare a fresh image drive from time to time.

Please advise!

pbatard, 2015-04-07:
> What would you recommend as start point if I were to rewrite the application for Linux and MacOSX.

I'd look at the Copyright notices (in the about box) to get the list of all the Open Source projects Rufus leverages, as most of these should have a Linux/OS X port and it should give you an idea of the pieces you're going to have to fill.

Then I'd try to produce a GUI tool that can format a drive to NTFS and produce a progress bar when doing so. Once you have that, you should be able to build upon it.

> Is it a huge hassle to handle block devices? should be much easier than on windows i guess.

I expect that to be easier. But Rufus does a lot of it's handling (boot records) through [ms-sys](http://ms-sys.sourceforge.net/), that has block device R/W capabilities, so you should be able to reuse that.

> Does such tool exist for Linux/Mac OS X already ?

I'm afraid not. Be aware that I estimate porting Rufus to be a __HUGE__ undertaking (something that will demand someone working __full time__ on it at the very least 3 months to complete). The application looks deceptively simple, but it is anything but, especially as it is pretty tied to using Windows APIs, so you would have a lot of work to do porting it.

> I have a problem creating Windows bootable USB flash drive and I am trapped in a useless loop of lots of work until I prepare a USB flash on my MacBook Pro.

Instead of porting Rufus, if your goal is to be able to work with Windows images, I think you should look into using [wimlib](https://sourceforge.net/projects/wimlib/) and build a commandline application around it that uses ms-sys to install the boot block and call on some NTFS or FAT32 formatting. Outside of Windows To Go, creating Windows bootable USB drives for anything > Windows Vista is pretty easy. All you need is to be able to format the drive to NTFS or FAT32 and install the standard Windows bootloader. Then you can just copy the files.
And if you need to do anything custom with the `install.wim`, you should be able to use wimlib. So really, I don't think you actually need to port Rufus. Just get your hands on ms-sys, wimlib and you should be able to achieve what you want from command line.

pbatard, 2015-04-11:
I will be closing this request now, as I provided an answer and this is not something I am planning to work on.

closed 2015-04-11 by pbatard