pbatard/rufus #487: Rufus 2.1.649 not downloading ldlinux.sys and ldlinux.bss files automatically.

Indigaz, 2015-04-15:
During a recent install, and attempt was made to burn an ISO image to a USB device.  I was prompted to download the ldlinux.sys and ldlinux.bss files to begin the process, however, Rufus never made the connection to http://rufus.akeo.ie/files/syslinux-6.03/ to retrieve these files and just hung the software.  I manually downloaded and moved them to the /rufus_files/ directory.  After files were manually downloaded and moved, all worked correctly.

pbatard, 2015-04-17:
I'm seeing plenty of downloads of the `ldlinux.sys` and `ldlinux.bss` from the server, and there was no downtime recently.
I'm afraid it's not a Rufus issue but a problem with your network connectivity.

Closing.

closed 2015-04-17 by pbatard

phipix01, 2015-08-18:
got the same issue for xubuntu14.10 32bit iso im on win8.1.
and im pretty sure my network is fine.
[image]

pbatard, 2015-08-18:
> and im pretty sure my network is fine.

And I'm pretty sure the error is on your side. I just tested the download a few minutes ago, and if there was a network issue, you'd have plenty of people reporting about it. The report from April 17th and your report are the only ones I have gotten so far (and Rufus gets downloaded more than 1.5 million times each month, so it would be pretty explicit if it was the server...)

phipix01, 2015-08-18:
allright so just for information, rufus was able to download the files on its own and add them to the usb drive .. no further steps needed.

pneumatics, 2016-02-15:
I got error 407 while trying to download under a HTTP proxy. Please provide a workaround.

jazbo1, 2016-12-09:
I have having the same issue. In C:/Owners/Downloads/syslinux-6.03/Idlinux.bss and idlinux.sys but each time when I go to run rufus when syslinux 6.03 starts I receive the same message that philJ0st shows in his screenshot. Please advise.

pbatard, 2016-12-09:
>  I receive the same message that philJ0st shows

You do realize that that message is just a __notice__ giving you the possibility not to connect to the internet to download files, in case you don't want to. It is __NOT__ an error.

What happens if you click _Yes_?

webgrrrl, 2017-02-04:
I face this same issue, where Rufus just tries to connect and displayed a failed connection error after a while. I noticed it created a rufus_files folder under the EXE file path where I saved Rufus. My workaround was to manually download both ldlinux.bss and ldlinux.sys in the newly created folder, and restarted.

So connecting via Rufus itself doesn't work, but accessing them manually via https://rufus.akeo.ie/files/syslinux-6.03/ does.

Timokou, 2017-03-06:
Hello webgrrrl,
I have the same issue too, would you please send me the steps to follow about that.?
I'm trying to make a usb bootable for lubuntu. Do you know if these two files are necessary when making a ISO image dvd  ?
Thanks in advance

webgrrrl, 2017-03-06:
@Timokou you must have both files before you can use Rufus to create the image. The steps are pretty straightforward:
1) Run Rufus to start the ISO image creation, and make sure you click Yes if it asks you to download the files.
2) When you encounter any error message, cancel the installation. By this time, you should see some new folders created by Rufus under the same location where you save your Rufus EXE installation file, including one called "rufus_files".
3) Go to https://rufus.akeo.ie/files/syslinux-6.03/ 
4) Download both ldlinux.bss and ldlinux.sys into the folder called syslinux-6.03 within your rufus_files folder.
5) Run Rufus and try to create the ISO image again.
-end-

Timokou, 2017-03-06:
Thanks for your answer. 
I will proceed as described. If you also have any info for my question about an iso dvs please let me know
-end-

Timokou, 2017-03-07:
I tried as described, but after waiting for 20 - 30 min procedure ended but instead of about 904 mb size for Lubuntu the total size was just 18 mb. So something goes wrong. Do i have to be connected to the internet during process ? Anyway, i did the same job with a Zorin os Iso with success at once,and without connection to the network. I don't know why this happens with Lubuntu. Any comments are welcome. I'll try again some other options.

Timokou, 2017-03-08:
I finally restarted the computer and changed the usb stick. And success !

mevsme, 2017-03-15:
I had the same issue until I ran Rufus with administrator rights.

Indigaz, 2017-04-20:
Y.

sawyerputnam, 2017-07-02:
These are all good methods, but check that you are getting the correct files. There is also an error for Rufus that says you need to download the files for the syslinux 6.04. make sure you go to https://rufus.akeo.ie/files/syslinux-6.04/ for the 6.04 message and put them in the right folder. You can also go to https://rufus.akeo.ie/files/ and navigate to the right directory for any error/download message.

Fataly-Insane, 2018-10-15:
So my network is not the issue, and I think akeo.ie is down, or am I crazy?

NKSHV, 2018-11-01:
Go to https://rufus.akeo.ie/files/  and it says "page not found"

pbatard, 2018-11-01:
The URL you need to use will be displayed in the log. It may be something like `https://rufus.ie/files/syslinux-6.03/ldlinux.sys` but that is completely dependent on the image you use.

Just try the URLs reported until you get valid download. Manual downloads of Syslinux or GRUB resources is not something I support. If you are connected to the internet, then these files should be downloaded automatically. If they are aren't then you'll need to sort our whatever networking issue you have as, unless the github servers are down (which is __exceedingly__ unlikely and would be reported as major news) the connectivity issue is 100% on your end.

ibuprophen1, 2018-11-02:
In addition to the https://rufus.akeo.ie/files/ link... 

A check on that URL will result in the following: 

Redirect to: https://rufus.ie/files
Problems found: Meta Refresh

I believe that the problem found isn't that the URL is bad in itself, it's more like (the best way I can explain it) the URL doesn't have a way to be displayed because it's not intended as such. 

It looks like the intent is only a URL path used only to download a file somewhere within the URL. 

As @pbatard had posted, if you use the URL he posted "https://rufus.ie/files/syslinux-6.03/ldlinux.sys", that will be a good download. 

If you take that URL and use "https://rufus.ie/files/syslinux-6.03/", the URL is good in itself but, isn't formatted/setup to be displayed as a directory. 

I hope that I had explained this okay via text... :-) 

~Ibuprophen

pbatard, 2018-11-03:
Again, the links work fine. It's just that directory listing (i.e. the ability to browse the content of the `files/` repository as if it was a folder) has been disabled and I am __not__ planning to re-enable it again because, as opposed to what was the case on the old server, I would need to create an individual page for each directory and it's just too tedious to maintain.

I am no longer providing help for people who want to use Rufus in unconnected mode, because, since online content __ALWAYS__ changes (and the issue you are facing is the exact consequence of that), it's just not worth trying to point people to such and such workaround, because it's going to change again a few months down the line (for instance I could point you to https://github.com/pbatard/pbatard.github.io/tree/master/files, but that too __IS__ going to move away in less than 2 months now), and when that happens, all we're going to get is a new stream of _"Link xyz is broken"_.

At this stage, even if it sounds harsh, I will therefore say this: If you plan to use Rufus offline, then you're on your own.

ali3nnn, 2018-11-12:
I solved the problem by downloading the latest rufus update, I have used 2.3v instead of 3.0.

Virtualburn, 2018-11-12:
> "Again, the links work fine. It's just that directory listing"

Downloaded the files from https://github.com/pbatard/pbatard.github.io/tree/master/files/syslinux-6.04 (in my case) an copied to 'C:\_tools\rufus_files\syslinux-6.04'

And it ran fine after having the same dialogue appear and no connected to server error.

Rufus was run with Admin and has full internet access.  Cheers.

Oluwaseyi14, 2018-11-13:
pls which directory on my pc will i put the files so that  rufus wont need to download it.

ibuprophen1, 2018-11-13:
@Oluwaseyi14, I believe that the answer to your question is outlined (hopefully) on the following link. 

https://github.com/pbatard/rufus/wiki/Usage-Notes

Good Luck! :-) 

~Ibuprophen

L-Johny, 2019-01-06:
> > "Again, the links work fine. It's just that directory listing"
> 
> Downloaded the files from https://github.com/pbatard/pbatard.github.io/tree/master/files/syslinux-6.04 (in my case) then copied to 'C:_tools\rufus_files\syslinux-6.04'
> 
> And it ran fine after having the same dialogue appear and no connected to server error.
> 
> Rufus was run with Admin and has full internet access. Cheers.
Why i can not find the files you listed?

d6tdev, 2019-01-16:
I got it to work by upgrading to a new version which was able to download files

beganovich, 2019-01-16:
Same issue right now, latest version of Rufus. Tried deleting the rufus_files, no luck. Checked firewall, everything works fine.

beganovich, 2019-01-16:
Windows 7, x86. Something changed in cryptography?

```
Downloading https://rufus.ie/files/syslinux-6.04/20181013/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - cancelling
```

pbatard, 2019-01-16:
>  Something changed in cryptography?

Yes. We're being hosted by GitHub now, and GitHub does not allow clients to connect if they use SSL versions that have been deprecated because they have major security vulnerabilities.

Obviously (and unlike browsers like Firefox or Chrome, which is part of the reason they end up being so bloated), Rufus does not embed its own custom version of an SSL library but instead uses the one provided natively by the OS. So, if you are using an OS that still relies on a outdated version of the SSL library, you're going to have an issue downloading anything. Note that you will see the __exact same issue__ if you try to access the links above with Internet Explorer (since IE also uses the native system libraries).

Now for the points that might be of interest to you:
- If you are using a vanilla version of Windows 7 where __all__ the system updates have been applied, then your SSL library should have been updated to use the more modern versions of the SSL protocol and you should not see this error. At least, this is what I have seen when testing on my side
- Alternatively, you may want to see [this](https://support.microsoft.com/en-us/help/3140245/update-to-enable-tls-1-1-and-tls-1-2-as-a-default-secure-protocols-in) or [this](https://github.com/rust-lang/cargo/issues/5066), that relate to the problem you are seeing.
- You should __always__ check what the log says, and then search for any message you think might be relevant in this issue tracker. If you had done that, then you would have found issue #1232.

krish1111, 2019-01-28:
> I'm seeing plenty of downloads of the `ldlinux.sys` and `ldlinux.bss` from the server, and there was no downtime recently.
> I'm afraid it's not a Rufus issue but a problem with your network connectivity.
> 
> Closing.

> During a recent install, an attempt was made to burn an ISO image to a USB device. I was prompted to download the ldlinux.sys and ldlinux.bss files to begin the process, however, Rufus never made the connection to http://rufus.akeo.ie/files/syslinux-6.03/ to retrieve these files and just hung the software. I manually downloaded and moved them to the /rufus_files/ directory. After files were manually downloaded and moved, all worked correctly.

hello,
which directory do i move

sangramadhikary, 2019-02-23:
I got a solution just update your Refuse software by official website and it will work correctly

matthack3, 2019-03-10:
solucion https://www.silentdragon.com/download/Rufus/rufus_files/syslinux-6.03/20171017/

pbatard, 2019-03-10:
Sorry @matthack3, I do not allow links to unofficial mirrors on this tracker. But you can find the same content to the one you pointed to at https://github.com/pbatard/rufus-web/tree/gh-pages/files (knowing that for folders that don't exist, such as `20171017` you just need to pick the `ldlinux.bss` and `ldlinux.sys` from the parent directory).

But, __REALLY__ the solution is:
1. Make sure your Windows system has had all the latest updates applied, especially if you are running Windows 7, so that you can connect through HTTPS to the official Rufus servers
2. Make sure you use the latest version of Rufus

That's all. If you do that, then you __don't__ have to copy files manually.

XIV91774, 2019-03-15:
Actually they're right, it's not a network problem, rufus.akeo.ie did move the syslinux 6.04 files, 301 moved permanently

doremaster, 2019-03-22:
On Windows 7, I couldn't get past the download step even when I hit "No". Ended up:

1) Went to home wiki page for syslinux https://wiki.syslinux.org/wiki/index.php?title=Download
2) Drilled down into testing folder and downloadables here https://mirrors.edge.kernel.org/pub/linux/utils/boot/syslinux/Testing/6.04/
3) Downloaded the syslinux-6.04-pre1.zip
4) Unzipped and pulled out ldlinux.bss and ldlinux.sys from syslinux-6.04-pre1.zip\bios\core\
5) Copied those two into my Rufus-v3.4\rufus_files\syslinux-6.04\pre1 directory
6) Re-ran Rufus fresh
7) Drank a beer

Cheers

Momazo7u7, 2019-03-22:
> 2\. rufus_files\syslinux-6.04\pre1

In my case the path was rufus_files\syslinux-6.04\20180907 when installing lubuntu . Thanks dude... you have saved me the hours that would take me to apply all the updates from windows update (wich i find unnecesary now that i'm switching to lubuntu)

Abrahiv, 2019-03-26:
> On Windows 7, I couldn't get past the download step even when I hit "No" (pretty sure this cancels the whole operation, not just skips the download step). Ended up:
> 
> 1. Went to home wiki page for syslinux https://wiki.syslinux.org/wiki/index.php?title=Download
> 2. Drilled down into testing folder and downloadables here https://mirrors.edge.kernel.org/pub/linux/utils/boot/syslinux/Testing/6.04/
> 3. Downloaded the syslinux-6.04-pre1.zip
> 4. Unzipped and pulled out ldlinux.bss and ldlinux.sys from syslinux-6.04-pre1.zip\bios\core\
> 5. Copied those two into my Rufus-v3.4\rufus_files\syslinux-6.04\pre1 directory
> 6. Re-ran Rufus fresh
> 7. Drank a beer
> 
> Cheers

Thanks Bro, working 100%