pbatard/rufus #489: Getting the  Error: Access to the device is denied when using rufus to create bootable USB flash drive

LiorNardea, 2015-04-18:
Hey guys,
I am using Rufus to create a bootable usb flash drive in order to install ubuntu Operating system on my computer.
I'm getting the following Error: Access to the device is denied.
Here is the log if someone knows what did I do wrong:

Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001AC380
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use /isolinux/isolinux.cfg for Syslinux
Detected Isolinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
Could not detect Grub version
ISO label: Ubuntu 14.04.2 LTS amd64
  Size: 1042120704 bytes
  Has a >64 chars filename: No
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.05)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: ubuntu-14.04.2-desktop-amd64 (1).iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7dc6f127-e51b-11e4-8279-8b088f0bca79}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001AC380
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7dc6f127-e51b-11e4-8279-8b088f0bca79}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001AC380
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7dc6f127-e51b-11e4-8279-8b088f0bca79}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001AC380
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
ABORTED: Cannot use an image that is located on the target drive!
\\?\Volume{7dc6f127-e51b-11e4-8279-8b088f0bca79}\ was already mounted as D:\
Re-mounted volume as 'D:' after error

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001AC380
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32014630912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-04-18:
Well, the log is pretty clear as to what your problem is:
```
ABORTED: Cannot use an image that is located on the target drive!
```
You are trying to reformat the drive that contains the ISO image you want to apply to it.
If Rufus was to allow that, it would delete that ISO right before it needs to access its content to copy files. You are sawing the branch you are sitting on!

Please move your ubuntu ISO __away__ from the drive you are trying to format (and understand that Rufus will erase EVERYTHING on that drive, so you will lose all your data there).

Closed as not a Rufus issue.

closed 2015-04-18 by pbatard

phantom131234, 2017-04-24:
and how do i move it elsewhere?I cant find a way to do it actually...

pbatard, 2017-04-24:
1. Please don't piggyback on __closed__ issues. Most of the time, you won't receive a response because the issue has already been closed.
2. Do you actually get the same error? i.e. are you really seeing `ABORTED: Cannot use an image that is located on the target drive!` in your log?. If not, this is __not__ the same issue at all.
3. If you do, you just need to use File Explorer to move the ISO from the USB drive to your computer's hard drive. If you don't know how to do that, then I will encourage you to watch tutorials on how to use Windows' File Explorer, as it is not the job of this issue tracker to teach you how to perform basic operations.