pbatard/rufus #493: Kaspersky Rescue Disk written, but with a message about do not support to the bootloader grub of Kaspersky Rescue Disk, and then fails to boot.

Axtish, 2015-04-19:
Kaspersky Rescue Disk written, but with a message about do not support to the bootloader grub of Kaspersky Rescue Disk, and then fails to boot. I would like to see Rufus supported Kaspersky Rescue Disk.

pbatard, 2015-04-19:
Duplicate of #467. I tried to see if I could add support Kaspersky, but they deliberately remove the version from the GRUB loader they use, which makes it impossible.

Please complain to Kaspersky, as these guys should no better that security through obscurity is no security at all.

Closing issue.

closed 2015-04-19 by pbatard

Axtish, 2015-04-19:
Kaspersky Rescue Disk works with WinSetupFromUSB 1.4. For him, I am using this program.

pbatard, 2015-04-19:
> Kaspersky Rescue Disk works with WinSetupFromUSB 1.4. For him, I am using this program.

Different applications create bootable USB in different manners. Some will copy ISOs wholesale (and use something like grub or Syslinux's ISO mode), which has both its advantages and drawbacks. Rufus isn't one of them.

The point is, Rufus could easily support Kaspersky, as it supports other GRUB based ISOs, if they didn't chose to deliberately hide the GRUB version they use. There's absolutely no reason to hide that version in the binary. But Kaspersky chose to hide it (and they had to modify the official GRUB source to do so).

Sopor-, 2015-04-19:
Or it is possible to use use Kaspersky's own program
http://rescuedisk.kaspersky-labs.com/rescuedisk/updatable/rescue2usb.exe

pbatard, 2015-04-19:
1. It's closed source.
2. I don't want to add something super specific just for Kaspersky.

If you are a Kaspersky customer, I would strongly advise you to contact them to ask them why they feel the need to use their own obfuscated version of GRUB, and whether their team of developer could try to use an official version, that doesn't hide its version, with the next release.