pbatard/rufus #494: Cannot create new bootable usb Unable to create file: [0x00000002]

Dmanisaround, 2015-04-23:
I'm using Rufus 2.1.649 and trying to create a bootable usb with an Adata uv128 /16 usb 3.0 drive. Windows 8.1

The process goes along normally until at the same point every time the usb port dies and losses the usb drive and then ( the usb drive is completely disconnected.) Here is the log. Note that it crashes on the same file every time Extracting: D:\lsu_lsi_.v00 (78 KB). Note that this does not happen when I format any of my usb 2.0 flash drives using the same iso.

Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'SDHC Card' (????:????)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15927345152 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15927345152 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB device 'SDHC Card' (????:????)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15927345152 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB device 'SDHC Card' (????:????)
Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15927345152 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Found USB device 'SDHC Card' (????:????)
Found USB 3.0 device 'ADATA USB Flash Drive USB Device' (125F:DB8A)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1936, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15927345152 bytes)
  Start Sector: 8192, Boot: No, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use /isolinux.cfg for Syslinux
Detected Isolinux version: 3.86 (from '/isolinux.bin')
ISO label: ESXI-6.0.0-2494585-STANDARD
[... 184 lines cut ...]
Extracting: D:\efi\boot\boot.cfg (1.6 KB)
Extracting: D:\efi\boot\bootia32.efi (94.2 KB)
Extracting: D:\efi\boot\bootx64.efi (96.3 KB)
Extracting: D:\efiboot.img (1 MB)
Extracting: D:\ehci_ehc.v00 (33.3 KB)
Extracting: D:\elxnet.v00 (118 KB)
Extracting: D:\emulex_e.v00 (5.7 KB)
Extracting: D:\esx_dvfi.v00 (89.2 KB)
Extracting: D:\ima_qla4.v00 (359.4 KB)
Extracting: D:\imgdb.tgz (40.4 KB)
Extracting: D:\imgpayld.tgz (3.9 MB)
Extracting: D:\ipmi_ipm.v00 (9.4 KB)
Extracting: D:\ipmi_ipm.v01 (23.0 KB)
Extracting: D:\ipmi_ipm.v02 (27.9 KB)
Extracting: D:\isolinux.bin (14 KB)
Extracting: D:\isolinux.cfg (267 bytes)
Extracting: D:\jumpstrt.gz (20 bytes)
Extracting: D:\k.b00 (3.2 MB)
Extracting: D:\lpfc.v00 (492.5 KB)
Extracting: D:\lsi_mr3.v00 (68.8 KB)
Extracting: D:\lsi_msgp.v00 (118.0 KB)
Extracting: D:\lsu_hp_h.v00 (15.3 KB)
Extracting: D:\lsu_lsi_.v00 (78 KB)
  Unable to create file: [0x00000002] The system cannot find the file specified.
Unable to create D:\syslinux.cfg - booting from USB will not work

pbatard, 2015-04-23:
> Note that this does not happen when I format any of my usb 2.0 flash drives using the same iso

The problem is clearly with your hardware, not Rufus.
If you plug your Adata device to an USB 2.0 port (or use an USB 2.0 extension cable), you will most likely find that it works just fine, which indicates that your issue is with high speed interferences, which can be due to poor cabling (are you using an USB 3.0 extension cable by any chance?) or poor controller isolation.

Rufus is usually much faster than other tools, so it tends to exacerbate these kind of hardware issues. Short of forcefully trying to slow down transfers, which is not something I want to do because I don't think people who can run properly at USB 3.0 speeds should be penalized by people using subpar hardware, this is not something that can be addressed in Rufus...

Please try on another computer that supports USB 3.0, and see what happens. Also it might help to check the SHA1 of your ISO just in case (though the fact that it works with an USB 2.0 device indicates that this is probably not a corrupted ISO issue).

For the record, I have had no issue using `VMware-VMvisor-Installer-6.0.0-2494585.x86_64.iso` (SHA1: `a38a9d37ea529329338de049679c1dd1687d3860`) with the fastest USB 3.0 device I have (Mushkin Ventura Ultra), even as I was using an USB 3.0 extension cable (The fact that it's [UAS](http://en.wikipedia.org/wiki/USB_Attached_SCSI) rather than USB shouldn't matter, as Rufus doesn't do anything different, and I also tried with a slightly slower but non UAS USB 3.0 device, with the same successful outcome):

```
Rufus version: 2.2.651
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000B8B04
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 111.8 GB (120033075200 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use '/isolinux.cfg' for Syslinux
Local time: 13074300996
Checking for Rufus updates...
Checking release channel...
Base update check: Rufus_win_x64_6.3.ver
Trying Rufus_win_x64_6.3.ver
Detected Isolinux version: 3.86 (from '/isolinux.bin')
ISO label: ESXI-6.0.0-2494585-STANDARD
  Size: 365021184 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (3.86)
    With an old menu.c32: Yes
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: VMware-VMvisor-Installer-6.0.0-2494585.x86_64.iso
Trying Rufus_win_x64_6.ver
Trying Rufus_win_x64.ver
Trying Rufus_win.ver
Found match for Rufus_win.ver on server http://rufus.akeo.ie/
Server time: 13074300996
Successfully downloaded version file (1060 bytes)
UPDATE DATA:
  version: 2.1.649 (release)
  platform_min: 5.1
  url: http://rufus.akeo.ie/downloads/rufus-2.1.exe
No new release version found.
Checking beta channel...
Base update check: Rufus_beta_win_x64_6.3.ver
Trying Rufus_beta_win_x64_6.3.ver
Trying Rufus_beta_win_x64_6.ver
Trying Rufus_beta_win_x64.ver
Trying Rufus_beta_win.ver
[... 174 lines cut ...]
Extracting: F:\vmware-esx-base-osl.txt (1.1 MB)
Extracting: F:\vmware-esx-base-readme (2.2 KB)
Extracting: F:\weaselin.t00 (1.9 MB)
Extracting: F:\xhci_xhc.v00 (57.1 KB)
Extracting: F:\xorg.v00 (1.3 MB)
Created: F:\syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Using autorun.inf label for drive F: 'ESXI-6.0.0-2494585-STANDARD'
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 14593, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00089D81
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 111.8 GB (120033075200 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

The issue is with your hardware...

Dmanisaround, 2015-04-23:
I'm using a microsoft surface pro connecting the usb drive directly to the usb 3.0 port. No hubs no cables no "subpar" hardware.

pbatard, 2015-04-23:
Then it's a SP incompatibility. As I explained, I see no issue whatsoever on my side, and I know a few other people have been able to use Rufus 2.x to install `VMware-VMvisor-Installer-6.0.0-2494585.x86_64.iso` on various USB devices.

Short of confirming that __ALL__ SP are affected, and someone sending me an SP to test, there's really nothing I can do.

pbatard, 2015-04-27:
Since there's nothing more I can do on this issue, I'm going to close it.

closed 2015-04-27 by pbatard

ypupo2002, 2015-05-08:
I have exactly this problem creating a Windows 8.1 bootable usb.

I'm using the same flash memory (Adata uv128 16GB usb3). Already tried two different memory sticks, on several machines (both sticks are brand new). The error occurs using both usb2 & usb3 ports, directly connected to the motherboard. It seems to be a problem with this specific brand and model.

deepakshrma, 2016-12-18:
1. Open Command Prompt, run as administrator mode
2. Open Rufus.exe

worked fine for me