pbatard/rufus #497: Unsupported ISO file from vmware "CPU Identification Utility"

lizard6, 2015-04-28:
Rufus is not be able to create a bootable disk using the ISO downloadable here : 
https://my.vmware.com/group/vmware/details?dlg=cpu_identification_utility&baseCode=ZHcqYmR0dGhidGR3#

pbatard, 2015-05-04:
I'll see what I can do.

lizard6, 2015-05-12:
Thank you in advance

        Sent by Outlook for Android

https://github.com/pbatard/rufus/issues/497#issuecomment-98692106

pbatard, 2015-05-24:
Unfortunately, while this ISO seems `Grub4DOS` based, it is missing `grldr` (which is why it isn't properly detected). So it probably uses some hybrid version of Grub 0.x, which trying to detect and add automatic support for (using Grub4DOS) looks like much more work than I'm willing to commit for an image that is unlikely to be used that much.

Note that you should might still be able to get that ISO to boot from USB by first installing `Grub4DOS` to your USB (click on the white triangle to get to advanced mode, which allows the installation of a bare `Grub4DOS`), and then copying the content of the ISO to the USB. But as with any boot process that was custom designed for an ISO, there's no guarantee that it will work (which is why I don't really want to go out of my way to try to support those).

I will close this issue now.

closed 2015-05-24 by pbatard