pbatard/rufus #500: error says drive is in use when using rufus

timj3371, 2015-05-05:
I fix it by running SDformat and then rufus 
Even right when create a drive i say lets create another one it still has error 
Here is the log 

Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003F02DC
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16007561216 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use /isolinux/isolinux.cfg for Syslinux
Detected Isolinux version: 4.05/20140113 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
Could not detect Grub version
ISO label: Ubuntu 14.04.2 LTS amd64
  Size: 1042120704 bytes
  Has a >64 chars filename: No
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (4.05)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: ubuntu-14.04.2-desktop-amd64.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
Re-mounted volume as 'G:' after error

IT happens with win 7 and win 8 and win 10 and linux 

Please help

pbatard, 2015-05-05:
Duplicate of #312.

Note that Rufus is not compatible with Linux, so I guess you tested using Wine, which is 100% __unsupported__.

Also, if I understand correctly, you do have a fix using SDFormat first.

Finally, you may want to check the [list of software Rufus is incompatible with](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with), as some of them will generate these kind of errors.

Closed as duplicate.

closed 2015-05-05 by pbatard

timj3371, 2015-05-05:
Pete it happens with any usb drive

It also happens with win 7 8 and 10

Thanks

pbatard, 2015-05-05:
Yes, I understand that. But if you actually read the link I pointed to, you would see that there is some software, such as _Defraggler_ or _Genie Timeline Backup Pro_ that keeps an exclusive access to the drive and will produce exactly the error you reported, for any USB drive you insert.

Some software (such as SDFormat, though I don't know enough about it to say that is the case) might try to ignore when other software has requested exclusive access to an USB drive. Rufus does not, because it considers that some software might have a very legitimate reason to want exclusive access, and therefore Rufus should not attempt to reformat it until access has been relinquished.

As per #312, I have some plans to try to report which application might be holding access to the drive and report it back to the user. But this is not a high priority issue for the time being compared to other ones. In case you are annoyed that it's unlikely that I'm going to have a chance to look into #312 for at least another few months, you may also want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

timj3371, 2015-05-05:
Ok I will look in to this

Thank you so much

timj3371, 2015-05-05:
Just solved it
I was using the front panel of my new case usb 3.0

Tried the back usb worked fine

Thanks
Tim

Alundra06, 2017-03-11:
@timj3371  suggestion is spot on, I changed the USB port and it worked for me as well.