pbatard/rufus #506: Error 0x80300024 when trying to install Windows 8.1 in unattended mode, using UEFI:NTFS

pbatard, 2015-05-13:
As reported by __deckman__ in [this reddit thread](http://www.reddit.com/r/techsupport/comments/35n052/is_there_a_way_to_do_a_uefi_windows_81_install/).

Tmp341, 2015-05-16:
There is a solution mentioned, says "Unplug all drives except the one you want to install on if you encounter error 0x80300024."
I think windows confuses itself, if there are more than one hdd.
Source : [Link](http://steffenwittig.com/2015/01/31/solved-installing-windows-8-1-fails-with-error-0x80300024/)

pbatard, 2015-05-16:
Yes, I mentioned that in the [reddit thread](http://www.reddit.com/r/techsupport/comments/35n052/is_there_a_way_to_do_a_uefi_windows_81_install/cr79vv4). However, I don't believe this was OP's case, especially as one would expect his FAT32 test to report the same code, which it didn't.

Quppa, 2016-03-30:
I experienced this issue yesterday when trying to install Windows 10 64-bit (1511 February 2016 refresh) to a new laptop with an NTFS-formatted USB drive. The drive configuration is (0) HGST HTS541010A7E630 and (1) NVMe Samsung SSD 950.

If the hard drive is plugged in and I try to install Windows to the SSD I get the 0x80300024 error. If I unplug the hard drive the error becomes 'Setup was unable to create a new system partition or locate an existing system partition'.

I tried modifying the boot order in the BIOS and I tried both the GPT and MBR partition schemes in Rufus before I thought to change the file system from NTFS to FAT32 (I've never previously had any issues with the former).

pbatard, 2016-03-30:
Okay, so if I understand correctly, you were trying to install Windows 10 to an internal Samsung SSD (did you try to clean that drive completely?) while having an HGST USB HDD plugged in, that was formatted as NTFS.

I think the most pressing issue is the `Setup was unable to create a new system partition or locate an existing system partition` error when the USB HDD is unplugged, which I understand only seems to happen if the USB installation flash drive was formatted as NTFS (I assume using _GPT for UEFI_ in Rufus, right?).
I'll try to see if I can replicate this issue, but I have a few questions before that:
1. Do you get the same issue with the non February Refresh of Windows 10 x64 1511?
2. Is your USB flash drive FIXED or REMOVABLE?
3. Even if the drive was created with the UEFI:NTFS partition, did you actually use UEFI:NTFS to boot or does the UEFI firmware of your target machine contain an NTFS driver (which seems to become more widespread these days), in which case you would had 2 boot options for the flash drive, and could have used the first one to bypass UEFI:NTFS?
4. Could I get the complete log of how you created your flash drive (when using NTFS)?

Quppa, 2016-03-30:
The two drives are both internal to the laptop (one NVMe SSD and one 2.5" HDD). The hard drive was formatted as NTFS (to be precise, there was also a 128 MB MSR partition on the hard drive which I deleted during setup).

The SSD started off with an MSR partition and a system recovery partition in addition to the main NTFS partition. I successfully deleted all the partitions from the drive within Windows setup - it's just that the next step (making new partitions) didn't work.

After formatting the USB flash drive as FAT32 I was able to proceed with the setup even with the internal hard drive plugged in.

2. The USB flash drive is fixed (it's a 64 GB Sandisk Extreme USB 3.0 drive). It shows up in the list of drives during Windows setup.

I'll have to do some more experimentation with the machine to give you proper answers to the other questions. I'm pretty sure I tried all three partition schemes, though (MBR partition scheme for BIOS/UEFI, MBR partition scheme for UEFI and GPT partition scheme for UEFI), and I don't believe the error message ever changed.

Quppa, 2016-03-30:
Something interesting to note: the recovery USB flash drive that came with the laptop is formatted as NTFS and doesn't have the same issue. In case it's relevant, it is a removable drive (8GB Kingston DataTraveler R3.0 G2).

pbatard, 2016-03-30:
Aha! That seems to confirm my suspicion that the Windows installer looks for all NTFS partitions on fixed drives, and treat them as either existing Windows installations, or possible targets for installation, which it is unhappy about...

If that's the case, I'm not sure what I can do in Rufus, as it's really an issue with the behaviour of the Windows installer (that installer is notorious in not liking what it considers as other OSes already installed, and Microsoft doesn't provide much detail on that process), short of advising to either use a removable drive, or, if using a fixed drive, make sure it is formatted as FAT32.

I still haven't had a chance to run much of a test, so I'll see what I can do, but I'm not very optimistic about tricking the installer into not mistaking its own partition as a target...

pbatard, 2016-03-30:
By the way, have you tried unplugging the installation drive before selecting the destination partition? This may throw a different error, as you proceed, but once you re-plug the install USB and click refresh, you may be able to proceed. I know [a similar trick](http://superuser.com/a/977181/286681) used to work for Windows 7.

Quppa, 2016-03-31:
I didn't try unplugging the installation drive, but I'll make a note to try that next time. Unfortunately I'm unable to experiment with the machine in question right now as I need it for work, but I should have an opportunity to play around with it again next week.

chrcoluk, 2016-04-28:
When I format my USB with rufus selecting GPT and NTFS, it doesnt make this small fat32 partition, is all 100% NTFS only.  What do I need to do to get this extra fat32 partition so it boots?

chrcoluk, 2016-04-28:
Right now I am copying the win files to the stick formatted as exfat hoping the bios can boot into exfat, as exfat seems the solution here since it supports large files and ntfs isnt a usb stick friendly filesystem anyway.

pbatard, 2016-04-28:
> What do I need to do to get this extra fat32 partition so it boots?

See the picture [here](https://github.com/pbatard/rufus/issues/615#issuecomment-215363905) if you want the blank UEFI:NTFS.

Or make sure you select a Windows ISO first and then set "GPT for UEFI" and "NTFS".

> hoping the bios can boot into exfat

I don't know of any BIOS or UEFI firmware that can boot natively into exFAT. But maybe you'll get lucky.

chrcoluk, 2016-04-28:
lol ok, doesnt seem obvious one should click the down arrow to make the option appear.

Sadly I dont need it now, I ended up splitting my wim file and am now using fat32.  But good to know incase I need in the future.

Seems daft the designers of UEFI made it only work with an obselete filesystem.

pbatard, 2017-04-19:
I don't think I'll ever get around to properly look into this. And since it should affect very few people &rarr; Closing.

closed 2017-04-19 by pbatard