pbatard/rufus #509: Unix/Linux Support

Aeirex, 2015-05-16:
Since Mac and Linux are pretty similar in nature, is there any plans to add in support for these operating systems? Unetbootin does not work very well on Mac OSX or Linux to be honest, and Ubuntu format tools are trash. Linux Mint has a good USB format tool built-in, but for newbies it is not practical for them to grab that package. It would be better for Rufus to be ported to these operating systems and be put in the stores.

pbatard, 2015-05-16:
As much as I'd like to do that, this is a major undertaking (__months__ of full time work), and since I can only develop Rufus in my __limited__ spare time, and there are still plenty of major features I want to add, that are of much higher priority than a port, that's not something I'm going to be able to do... if ever.

Then again, Rufus is Open Source, so if anybody wants to port it, they're more than welcome to do so.

Note that this question gets asked on a regular basis, so much so that [there is a FAQ entry for it](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS).

closed 2015-05-16 by pbatard

Aeirex, 2015-05-16:
No problem man, I was just curious. I may work on a fork of it if I get motivated to learn how to program more in C/C++.