pbatard/rufus #511: USB not recognized in Rufus. Read/Write working flawlessly

kekate, 2015-05-18:
R ufus is not detecting my device. I tried with another usb stick and it worked. I noticed when I plug in a working USB flash drive then a driver called something with the words of Mass storage appears in Device Manager, but when I plug in the drive which isn't  working then nothing new appears. I could read and write without any issues. Rufus log shows nothing, but 0 devices found. I've tried e everything. Formay, assign a new letter etc etc. Is my usb drive dead?

pbatard, 2015-05-18:
Have you tried the steps detailed in the [FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected)?

kekate, 2015-05-18:
Yes, but I still get a message that no devices were found. I found a program called WintoBootic and I think it's working right now. I'm flashing WIN7 iso. Hopefully WintoBootic reverts my USB flash drive back to its original state so I could make an UEFI flash drive for windows 7. I'll let you know. Thanks for answering by the way.

kekate, 2015-05-18:
Flashed WIN7 with WIntoBootic, but my device is still not detected by Rufus.

pbatard, 2015-05-18:
Then please provide a [full log](https://github.com/pbatard/rufus/wiki/FAQ#I_contacted_you_but_all_I_got_was_a_link_to_this__why). I know you mention that the log shows nothing, but I still would like to see it.

pbatard, 2015-05-23:
Again, I'd be interested in a full log. Also, is there any chance you could be using ASUS USB 3.0 Boost, with "Turbo Mode" enabled in your BIOS? I would also be interested about your USB details, such as the brand, capacity, type (USB 2.0, USB 3.0) and __exact__ model.

pbatard, 2015-05-28:
No update &rarr; Closing issue.

closed 2015-05-28 by pbatard