pbatard/rufus #513: Rufus can't find my two sandisk usb's

NaoDarkness, 2015-05-20:
Two USB's I just bought can't be found.

2X SanDisk Extreme CZ80 16GB USB 3.0
http://www.amazon.com/gp/product/B00KT7DOR0/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

Walkman100, 2015-05-20:
Can you see them in Windows Explorer? (File Explorer on Win8)

NaoDarkness, 2015-05-20:
Yes, I can see them both.

NaoDarkness, 2015-05-20:
Here Rufus's Log:

/////////////////////////////////////////////////////////////////////////////////////
Rufus version: 2.1.649
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:6545)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1948, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0C8C3188
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16025386496 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'PNY USB 2.0 FD USB Device' (154B:6545)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1948, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0C8C3188
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16025386496 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
0 devices found
0 devices found
/////////////////////////////////////////////////////////////////////////////////////
PNY USB was a test to see if the USB ports or rufus itself was not working.

pbatard, 2015-05-20:
What's your `E:` drive? Is it one of the SanDisk drives.
Also, you wouldn't have configured these drives for redundancy (RAID) by any chance?

NaoDarkness, 2015-05-21:
My E: drive is two 2TB HDD in raid 0. And no, the USB's are not in a RAID, If that was what you were asking. I have some USB 2.0 port in the back That I could try, I'll do it right now.

NaoDarkness, 2015-05-21:
I tried the USB 2.0 in the back they did not work.

But I figured out what was causing it, I have ASUS motherboard and there is a program called 'AiSuite 3'. This program allows you to monitor system settings, voltage, fan speed and to overclock without going to the BIOS, anyways, there is a feature in there called 'USB 3.0 Boost', which does as the name implies, Boost you USB 3.0 devices. 

But I wonder why 'AiSuite 3' was messing with rufus this time? I have another pair of USB 3.0 stick, dongles really, and they didn't give me an issue using rufus with USB 3.0 boost on.

pbatard, 2015-05-21:
>  there is a feature in there called 'USB 3.0 Boost', which does as the name implies, Boost you USB 3.0 devices.

Aha. Looking at [ASUS info for it](http://event.asus.com/mb/2010/The_Best_USB3_Experience/The_UASP_For_USB3.0.htm), it looks like, when running in "Turbo Mode" ASUS added their own sauce on top of standard UASP (which Rufus should be compatible with), to force a nonstandard feature. I suspect that they create a virtual device to add that feature, which of course means that they are likely hiding the actual device from the system.

I believe only "Turbo Mode" should be causing this issue, so I've added a [note about this in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with). If you have a chance, can you confirm that it's only the "Turbo Mode" that will cause the issue (i.e. "UASP Mode" and "Normal Mode" are okay)?

NaoDarkness, 2015-05-24:
Yes "Turbo Mode" was only thing stopping the USB's from being seen by rufus, the work just fine in "Normal Mode". As for "UASP Mode" I would have to check, I have a few HDD and a SSD or two, and see if rufus can still find them. 

But as for the USB's 3.0 I have, I don't think they're "UASP" capable, I would have to check.

And for a full log, are you talking about another rufus log or a full system log?

I don't think "Turbo Mode" is enabled in the BIOS, but i could check and see if there is an option for that. But I think "USB Boost" is a software implementation, unless ASUS is doing this through the hardware on the board.

NaoDarkness, 2015-05-24:
I just checked and "UASP" or "UASP Mode" works with rufus just fine. I don't know if "UASP" is running normally or ASUS is using like you said "add their own sauce on top of standard UASP". The reason I'm confused is that when I check in "AiSuite" UASP is already enabled, so I don't know if that by system default, ASUS doing or maybe It has to do with the cable I'm using? But anyways, everything seems in order.

http://www.startech.com/HDD/Adapters/USB-3-SATA-adapter-cable-with-UASP~USB3S2SAT3CB
USB 3.0 to 2.5” SATA III Hard Drive Adapter Cable w/ UASP – for SSD / HDD
Product ID: USB3S2SAT3CB

Still doing some work on the computer right now so can't go into BIOS right now though, maybe tomorrow?

Oh and the USB"s I'm using are:

http://www.sandisk.com/products/usb/drives/extreme/
2x SanDisk Extreme® USB 3.0 Flash Drive
16GB
USB 3.0
SDCZ80-016G

pbatard, 2015-05-24:
Thanks for the additional tests.

> I don't know if "UASP" is running normally or ASUS is using like you said "add their own sauce on top of standard UASP"

`UASP` is something that standard USB 3.0 controllers should provide, and that is defined in the USB specs, so this is the normal mode. So I guess what ASUS does is allow users the ability to disable that functionality altogether (which is OK - an USB device can run just fine without UASP) or enhance it with the `Turbo Mode`. The problem however is that the _enhancements_ ASUS applies are of their own doing and not in any standards, so they are probably very custom.

Still, I'd like to see if there's a way to get USB devices listed in Rufus when `Turbo Mode` is enabled. One possibility is that when that is activated, the USB disks are no longer listed as USB devices but as regular disk drives, which File Explorer won't care about, but which Rufus does (since it doesn't list physical internal disks by default, to avoid any possible data loss from users formatting a drive by mistake).

However, maybe I can use a similar trick to what I use to support Virtual HDs, by allowing drives with specific Hardware IDs, if Turbo Mode enabled USBs can be differentiated like this.

Can I ask you to download and run `device.exe` from [here](http://erwan.labalec.fr/other/devices.zip) (I didn't write that, but I found it useful when [troubleshooting VHD support](http://reboot.pro/topic/20178-specific-vhd-creation-and-mounting-problems/page-3)), select `DiskDrive` for Class and locate the info for your SanDisk Extreme when `Turbo Mode` is enabled.

It should look something like this, and I'm hoping the Hardware ID will be distinct enough from standard internal drives, to add support for it:
[image]

NaoDarkness, 2015-05-24:
I ran "device.exe" and this is what I got:
[image]
[image]
 I included both USB's just in case.

pbatard, 2015-05-24:
Thanks. I guess I can't use the `HARDWAREID` after all. But that's also true from drives in UASP mode.

Can I ask you to check another thing then? Can you please go to the Windows Device Manager, then under `Disk Drive` select your device, right click on it and select `Properties`, then go to the `Details` tab, and choose the `Device Stack` property. See screenshot below (where I used a Mushkin drive, as it uses UASP). 

This should list the names of the driver being used, which is what Rufus uses to try to identify UASP drives. The default UASP driver from Microsoft is `UASPStor` but I guess that, in `Turbo Mode`, this might be different. If that is the case, I could add what you have to the list of UASP drivers that Rufus knows, which would allow your device to be recognized.

[image]

NaoDarkness, 2015-05-25:
You were right, they are listed as something different, Here:
[image]

pbatard, 2015-05-25:
Great!

Hopefully, checking against `ASUSstpt` is all I need to add to get ASUS Turbo Mode supported.

I now have a test version of Rufus [HERE](https://rufus.akeo.ie/testing/), that adds the check for `ASUSstpt`, if you want to give it a try.

NaoDarkness, 2015-05-26:
I ran the test version and the USB's still are not visible, here is the log:

Rufus version: 2.2.665 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764': [0x00000057] The parameter is incorrect.
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011122140001370765': [0x00000057] The parameter is incorrect.
0 devices found

pbatard, 2015-05-26:
OK. Can you try the second TEST version from [here](https://rufus.akeo.ie/testing/)?
Regardless of whether that version works, I'm especially interested in the log you will get.

Also, can you provide a screenshot of how your device is seen in Device Manager when you list devices by connection? Basically, you need to select `View` &rarr; `Devices by connection` and then explore the tree until you find your device, like this:
[image]
According to your log, it seems that we can't get the USB information from your device when running in `Turbo Mode` so I am curious to see if Windows still sees it as a USB device.

NaoDarkness, 2015-05-27:
It works!!!:
[image]
[image]

Here's rufus log:

Rufus version: 2.2.665 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
devid_list[0] = ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764
is_ASUS = 1
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011122140001370765': [0x00000057] The parameter is incorrect.
Found ASUS 'Turbo Mode' USB device 'SanDisk Extreme SCSI Disk Device'
props.speed = 0, VID=0000, PID=0000
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
is_ASUS = 1
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764': [0x00000057] The parameter is incorrect.
Found ASUS 'Turbo Mode' USB device 'SanDisk Extreme SCSI Disk Device'
props.speed = 0, VID=0000, PID=0000
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
2 devices found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16008593408 bytes)
  Start Sector: 32, Boot: No, Recognized: Yes
devid_list[0] = ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011122140001370765
is_ASUS = 1
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011122140001370765': [0x00000057] The parameter is incorrect.
Found ASUS 'Turbo Mode' USB device 'SanDisk Extreme SCSI Disk Device'
props.speed = 0, VID=0000, PID=0000
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
1 device found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16008593408 bytes)
  Start Sector: 32, Boot: No, Recognized: Yes
devid_list[0] = ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764
is_ASUS = 1
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011122140001370765': [0x00000057] The parameter is incorrect.
Found ASUS 'Turbo Mode' USB device 'SanDisk Extreme SCSI Disk Device'
props.speed = 0, VID=0000, PID=0000
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
is_ASUS = 1
Could not get node connection information for 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764': [0x00000057] The parameter is incorrect.
Found ASUS 'Turbo Mode' USB device 'SanDisk Extreme SCSI Disk Device'
props.speed = 0, VID=0000, PID=0000
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
2 devices found
No volume information for drive 0x84
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
[... 1713 lines cut ...]
Extracting: F:\sources\sxs\x86_netfx-aspnet_wp_exe_b03f5f7f11d50a3a_6.3.9600.16384_none_dff60f8750549d0e\aspnet_wp.exe (30.6 KB)
Extracting: F:\sources\sxs\x86_netfx-assemblylist_xml_b03f5f7f11d50a3a_6.3.9600.16384_none_71ff1d80086a7d54\frameworklist.xml (10.9 KB)
Extracting: F:\sources\sxs\x86_netfx-clr_ilasm_exe_b03f5f7f11d50a3a_6.3.9600.16384_none_1e1d5e9c51d9d1dd\ilasm.exe (223.1 KB)
Extracting: F:\sources\sxs\x86_netfx-clr_ilasm_exe_b03f5f7f11d50a3a_6.3.9600.16384_none_1e1d5e9c51d9d1dd\ilasm.exe.config (181 bytes)
Extracting: F:\sources\sxs\x86_netfx-clr_mof_b03f5f7f11d50a3a_6.3.9600.16384_none_68f887910edcac78\clr.mof (21.5 KB)
Extracting: F:\sources\sxs\x86_netfx-clr_mof_b03f5f7f11d50a3a_6.3.9600.16384_none_68f887910edcac78\clr.mof.uninstall (907 bytes)
Extracting: F:\sources\sxs\x86_netfx-clr_sys_entservcs_thunk_dll_b03f5f7f11d50a3a_6.3.9600.16384_none_6d6d41137be5c295\system.enterpriseservices.thunk.dll (53.7 KB)
Extracting: F:\sources\sxs\x86_netfx-config_files_.._gacutil_exe_config_31bf3856ad364e35_6.3.9600.16384_none_319d471cf1e70c25\gacutil.exe.config (140 bytes)
Extracting: F:\sources\sxs\x86_netfx-config_files_.._regsvcs_exe_config_31bf3856ad364e35_6.3.9600.16384_none_ee1123c8a4553b95\regsvcs.exe.config (353 bytes)
Extracting: F:\sources\sxs\x86_netfx-corperfmonext_dll_b03f5f7f11d50a3a_6.3.9600.16384_none_cdf4cb1286ff751f\corperfmonext.dll (85.7 KB)
Extracting: F:\sources\sxs\x86_netfx-csc_exe_config_b03f5f7f11d50a3a_6.3.9600.16384_none_a2814c7f5d1c2211\csc.exe.config (221 bytes)
Extracting: F:\sources\sxs\x86_netfx-csharpcompilermsg_b03f5f7f11d50a3a_6.3.9600.16384_none_8be1632aaee2b7ef\cscompui.dll (139.6 KB)
Extracting: F:\sources\sxs\x86_netfx-csharp_compiler_cscomp_b03f5f7f11d50a3a_6.3.9600.16384_none_447a5af895ee0657\cscomp.dll (1.1 MB)
Extracting: F:\sources\sxs\x86_netfx-csharp_compiler_csc_b03f5f7f11d50a3a_6.3.9600.16384_none_1985dc1cf125db97\csc.exe (76.1 KB)
Extracting: F:\sources\sxs\x86_netfx-csharp_compiler_csc_b03f5f7f11d50a3a_6.3.9600.16384_none_1985dc1cf125db97\csc.rsp (976 bytes)
[... cut ...]
Extracting: F:\sources\sxs\x86_netfx-dv_aspnetmmc_chm_res_b03f5f7f11d50a3a_6.3.9600.16384_none_0f83ef8eda3a7e3d\dv_aspnetmmc.chm (112.4 KB)
Extracting: F:\sources\sxs\x86_netfx-dw_b03f5f7f11d50a3a_6.3.9600.16384_none_e8a9a77fdf448e75\dw20.exe (33.1 KB)
Extracting: F:\sources\sxs\x86_netfx-eventlogmessages_dll_b03f5f7f11d50a3a_6.3.9600.16384_none_ea719569b704c681\eventlogmessages.dll (777.2 KB)
Extracting: F:\sources\sxs\x86_netfx-fusion_dll_b03f5f7f11d50a3a_6.3.9600.16384_none_1dde9c89cea5b4f1\fusion.dll (16.1 KB)
Extracting: F:\sources\sxs\x86_netfx-fw_netfxperf_dll_31bf3856ad364e35_6.3.9600.16384_none_29336d0996d2a40e

pbatard, 2015-05-27:
Thanks for the new test.

I need to think about the information you gave and decide what I want to do. That extra `ASUS USB 3.0 Boost Storage Driver` could be an annoyance, and I don't like the fact that Rufus can't get the `VID` and `PID` of these ASUS "boosted" USB devices, especially if autodetection seems to be hit or miss...

Also, can you please avoid testing with __both__ of your SanDisk USB drives connected for the time being (as in your screenshot)? I'd rather not add the special case of having 2 devices of the exact same type connected to the same hub, while trying to figure out how to handle this whole ASUS mess, as it makes it more difficult to interpret the data.

Finally, can I ask you for one more thing right now? Can you right click on the `ASUS USB 3.0 Boost Storage Driver` in Device manager, that sits on top of your SanDisk USB device, select `Properties`, then go to the `Details` tab, and report the `Hardware Ids` value?

NaoDarkness, 2015-05-27:
It seems to be autodetecting now, rebooted a few times, tried USB port on the front and back.
I really think it was something that was wrong on my side, not rufus.

Here's the Hardware Id:
[image]

pbatard, 2015-05-27:
Thanks.

I think I see a bit clearer now, and I may have a way to get your device properly listed with `PID`:`VID` and all. I also identified an bug in my code, which didn't help with the detection process.

Can you please try [TEST3](https://rufus.akeo.ie/testing/) and let me know how that works?
At the very least, his version should generate a bunch of extra messages in the log, which I'm interested in...

NaoDarkness, 2015-05-28:
Ran TEST3 and it found the USB right away, did a test with a windows ISO to be sure, everything went fine.

Here's rufus log:

Rufus version: 2.2.665 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
STORING[158] = 1: 'USB\VID_8087&PID_8008\5&49593B2&0&1'
STORING[60] = 2: 'USB\VID_8087&PID_8000\5&18088FC2&0&1'
STORING[109] = 3: 'USB\VID_2109&PID_0812\6&13A1515A&0&4'
STORING[28] = 3: 'VID_0781&PID_5580&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\AA011121142358540764'
STORING[169] = 4: 'USB\VID_174C&PID_3074\5&15C311E1&0&18'
STORING[83] = 4: 'USB\VID_1532&PID_011A\5&15C311E1&0&14'
STORING[202] = 4: 'USB\VID_1B1C&PID_0C04\5&15C311E1&0&7'
STORING[141] = 4: 'USB\VID_1532&PID_0037\5&15C311E1&0&5'
STORING[256] = 4: 'USB\VID_174C&PID_2074\5&15C311E1&0&3'
STORING[7] = 4: 'USB\VID_2109&PID_2812\5&15C311E1&0&9'
STORING[156] = 4: 'USB\VID_2109&PID_0812\5&15C311E1&0&20'
STORING[159] = 5: 'USB\VID_2109&PID_2812\6&1949A51&0&4'
devid_list[0] = ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764
197 = 'ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\AA011121142358540764', data=0
TRY AGAIN: 28 = 'VID_0781&PID_5580&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\AA011121142358540764', data=3
'\\?\usb#vid_2109&pid_0812#5&15c311e1&0&20#{f18a0e88-c30c-11d0-8815-00a0c906bed8}' -> 'VID_0781&PID_5580&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\AA011121142358540764'
GOT NODE INF FOR: 'VID_0781&PID_5580&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\AA011121142358540764'
Found UAS (USB 3.0) device 'SanDisk Extreme SCSI Disk Device' (0781:5580)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0D8313CD
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-05-28:
Great! Then I think we have a winner. :smile: 

I have applied the changes to the latest BETA (BETA 2, which can be found [here](https://rufus.akeo.ie/downloads/). Can you please test it and confirm that it works there too?

This issue will be automatically closed when I push the relevant commit, but feel free to reopen/add on to to it if you still see problems.

closed 2015-05-28 by pbatard

NaoDarkness, 2015-05-28:
Tested with ```BETA 2```, everything is working great!

Here's the log:

```
Rufus version: 2.2.666 (Beta)
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found UAS (USB 3.0) device 'SanDisk Extreme SCSI Disk Device' (0781:5580) [GP]
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
Ignoring drive '\\.\E:' as it spans multiple disks (RAID?)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0D8313CD
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

You don't know how happy I was to be of some help, been using rufus for the past two years for fixing computer and builds, never thought I would contribute to such a great program.

Keep up the awesome work!:smile:

pbatard, 2015-05-28:
Thanks for confirming that the BETA looks good.

Besides adding support for "Turbo Mode", your issue helped fix a potentially nasty bug, so I __really__ appreciate the time you took for all the testing and reporting - couldn't have done it without your help!

commit ac213a16 (2015-05-28): [core] fix an enumeration issue and add support for ASUS UASP Turbo Mode

* Our detection for a match in the hash table, during USB enumeration, always
  pointed to the first hub when no match was found.
* Also, some USB mass storage devices (ASUS Turbo Mode devices, iPods) may
  have an additional driver inserted, which we now try to handle by looking
  up the grandparent.
* Thanks to NaoDarkness for helping investigate this
* Closes #513