pbatard/rufus #514: AVG Internet Security

ahmedflix25, 2015-05-20:
i have to temporarily disable it because rufus tells me that my usb used by another procces although avg don't scan it or anything even some times i have to end it at all so Rufus can format and burn the iso

pbatard, 2015-05-20:
Rufus' policy is not to take control of a drive if an application has requested exclusive access to it, even if they don't do anything (because it's bad practice and could be all the difference it takes to avoid data loss).

It is a well known issue that __some__ security applications tend to request exclusive access to drives when they shouldn't. See [this list from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with). This is a pure security application issue.

Now, some vendors (e.g. ESET Smart Security, which used to be in that list) will acknowledge the issue and fix it. Others don't.

As there is nothing for me to do, I will close this issue. You will need to contact AVG and ask them to fix their software.

closed 2015-05-20 by pbatard

Tmp341, 2015-05-21:
When you see that error, try looking the drive with "Unlocker"

ahmedflix25, 2015-05-22:
ok, thanks