pbatard/rufus #515: Running from Linux?

Morketh, 2015-05-22:
Was wondering how difficult it would be to port this utility to a *ix like platform. Im currantly running bolth Redhat Enterprise and Debian, ive used this tool a thousand times from windows 7 and im very pleased with the work would love to add this to my Linux/Unix tool kit esseptialy on a LiveCD this would come in very handy for one of those times you have a USB stick internet access and only a LiveCD at your hands for on-the-fly diagnostics.
thanks in advance

Halamix2, 2015-05-22:
See #509 , #485 and #93
BTW maybe it's good idea to keep one issue opened about Linux to prevent asking 4 times about the same.

pbatard, 2015-05-22:
My estimate is that it would probably take 2 months __full time__ work to port Rufus to Linux.

As to keeping one issue open, I don't see it necessary as the only people who request a Linux port in the issue tracker are people who [didn't read the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS). I also only keep an issue open if I plan to work on it at some stage... :smiley:

closed 2015-05-22 by pbatard

Morketh, 2015-05-22:
Thank you for the information and the fast responses. I'll hav a look a the FAQs must have missed those while i was looking for a solution.