pbatard/rufus #517: bcryptprimitives.dll error

ressing1, 2015-05-28:
I am trying to create a windows 10 to go build 10074 using Windows 8.1 x64.  Every time I attempt this I get the following error:

[path]\bcryptprimitives.dll is either not designed to run on Windows or it contains an error. Try installing the program again using the original installation media
[image]

pbatard, 2015-05-28:
Please see the [FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Windows_To_Go_issues).

Currently, you can only create Windows To Go image from a version of Windows that is the same or more recent than your target (which means you should only try to create Windows 10 To Go from Windows 10 or higher).

And yes (#453) I will be trying to improve that on a future version of Rufus (but these things take time).

Also I have personally not had any success getting a Windows 10 x64 To Go image working, even when using the official Microsoft tool...

I will close this issue now.

closed 2015-05-28 by pbatard

ressing1, 2015-05-28:
I have being using Windows 10 to go as my daily OS.  I use rufus to create
the drive.  When I get the bcryptPrimitives.dll error, I End Task the
bcdboot.exe process and run "bcdboot f:\windows /s f:" from command line.
This works fine and the Windows 10 boots perfectly.  I think windows 10 to
go is pickier about what type of drive you use.  I am using a 128GB
mydigitalSSD drive which is a USB 3  msata enclosure which works
perfectly.  I have had limited sucess making windows 10 to go with Flash
Drives.  Rufus is an excellent tool. Thank you.

pbatard, 2015-05-28:
As I said, I'll see what I can do. As long as Windows 10 hasn't been officially released, and Microsoft are free to overhaul their To Go set up as they see fit (again, from what I could see with the build I try, even their official To Go creation tool seemed to have issues), Windows 10 support is 100% experimental.

ressing1, 2015-05-28:
I understand that.  I want trying to be critical. I just wanted to let you
know about my workaround in case that helps. I completely understand that
windows 10 is an unknown quantity. I think Rufus is a great tool and just
wanted to help.